jayofelony/pwnagotchi #146: [HELP] starting monitor interface

bulkpet, 2024-04-28:
Hi all, writing this because I'm not finding nor understanding the situation; I'm using last version of pwnagotchi 2.8.9. Everything it's correct when i flash it and use it normally.

The problems starts happening when i disable pwnagotchi, bettercap and pwngrid services; if i try to run `start_monitor_interface` from  `/usr/bin/pwnlib` i get `operation not supported -95`.

Then i try to run `reload_brcm` but i get that the module is in use.

So i try to run `lsmod  |grep brcmfmac` and the result are this

```
brcmfmac_wcc           12288  0
brcmfmac              348160  1 brcmfmac_wcc
brcmutil               24576  1 brcmfmac
cfg80211              995328  1 brcmfmac
```

Not even a reboot solves the issue, what could be the problem? I'm running RPi 4b.

jayofelony, 2024-04-28:
If you stop pwnagotchi, monitor mode is still active. So you don't need to run that command.

bulkpet, 2024-04-28:
Yes, yes in fact i try to run it only when i have no wlan0mon interface. 

Just reflashed, restored a previous 2.8.9 working backup, rebooted and wlan0mon disappeared; and no matter what I try of the above, i cannot get wlan0 to be in monitor mode again.

closed 2024-04-28 by bulkpet