jayofelony/pwnagotchi #151: Screen get wiped when shutting Down

JAKAMI99, 2024-05-06:
**Describe the bug**
Screen get wiped when shutting down.
The "Good Night" message works, but afterwards the screen seems to get cleared by some kind of function.

**Expected behavior**
The "Good Night - ZzZzZ" message appears and stays instead of a plain white screen.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version
 - OS version 2.9.1
 - Type of hardware Zero 2

JAKAMI99, 2024-05-06:
The ui.invert option does not change the behavior. 
The Screen is always white, really weird.

JAKAMI99, 2024-05-06:
I checked a few things, but no luck.
It happens no matter what I do. I tried to shut down the pwnagotchi with the web interface over my computer and also with my phone over Bluetooth tethering. 
It always does a screen clear by flickering a few times after the "Zzz" face is shown for about 6 seconds.

jayofelony, 2024-05-06:
personality.clear_on_exit = true # clear display when shutting down cleanly

JAKAMI99, 2024-05-07:
Thanks!
 But why would you implement it, is there a reason and why is it enabled by default?

closed 2024-05-07 by JAKAMI99