jayofelony/pwnagotchi #155: "e-Paper busy"

iandouglas, 2024-05-16:
Raspberry Pi Zero 2 W, happens on both this build and on aluminum-ice. Posting here per @jayofelony's direction on https://github.com/waveshareteam/e-Paper/issues/339

I've copied @jayofelony's epdconfig.py file and still get "e-Paper Busy" on several different screens.

Since I don't always know which script to run based on the hardware, I'm looping over all of the scripts for that screen size like

for i in *2in13*py; do python3 $i; done

but they're all telling me that the e-Paper is busy.

I've tried one 2.13" screen, which is marked as a v4 but was used in a project (pwnagotchi) as a v3, but now seems to not refresh any more. I've also tried two separate 2.7" screens with the 4 buttons, as well as a large 7.5" display.

I've seen in the open issues and closed issues that the C examples might work better to clear the busy bit, but I can't compile any of the scripts because of a missing "lgpio.h" file ... I've also tried the other three settings in the Makefile for USE_LIBRPI but I get an error of missing header files for all 4 of the options.

I suspect there's something wrong with my RPi Zero 2 W maybe, but would like help diagnosing.

For config.toml the 2.13" was originally set up as a "waveshare_3" but the display is labeled with a "V4" sticker on the back, so I changed it to "waveshare_4" and it worked for a while, but I changed it back to "waveshare_3". Shortly after, the screen stopped updating. I suspected my pisugar3 battery as it was acting flaky, but removing that didn't help the screen update any more.

I swapped the 2.13" screen for a [2.7-inch waveshare display](https://www.waveshare.com/product/displays/e-paper/epaper-2/2.7inch-e-paper-hat.htm) with 4 buttons which didn't work at all, I had it configured as a "waveshare2in7b_v2", but pwnagotchi and the waveshare e-paper example code for all of their 2.7-inch screens did nothing to draw on the screen, they all report "e-Paper busy".

So I swapped to a [massive 7" waveshare screen](https://www.waveshare.com/product/displays/e-paper/epaper-1/7.5inch-e-paper-hat-b.htm), but same as the 2.13 and 2.7 ... all I get is "e-Paper busy" and no display. I tried each of the "waveshare7in5*" configurations and got nothing, and the Python scripts were also "e-Paper busy".

I've got two new RPi Zero 2 W's coming in over the next few days to help rule out if it's the Pi 'cause at this point 4 screens being bad on the same Pi seems unlikely (all except the original 2.13" were new out of box within the past week). I have one headless pwnagotchi here at home but it doesn't have header pins.

What would be helpful from the logs? I don't generally see any errors in the pwnagotchi.log at all, just that nothing gets drawn on the screen. The only time I see an "e-Paper busy" is when I'm running Waveshare's example Python scripts, and I run those like this:

```bash
for i in *2in7*py; do sudo killall pwnagotchi bettercap pwngrid ; echo $i; echo; echo; python $i; echo; echo; echo; done
```
that way it tries to kill all of the pwnagotchi scripts that might be trying to use the screen in case it's a problem with sharing the I2C device across applications or something.

jayofelony, 2024-05-16:
What epdconfig did you copy btw? I have a working one in 2.8.9

iandouglas, 2024-05-16:
@jayofelony I copied over the one that you were recommending to people back in January/February in the waveshare examples repo. eg https://github.com/waveshareteam/e-Paper/issues/321#issuecomment-1890064586

iandouglas, 2024-05-16:
I'll try copying the one from your 2.8.9 build to the directory where these other examples live from the waveshare team to see what's up. I still suspect it's my Pi02W acting up... New Pi's should be here today/tomorrow as well.

jayofelony, 2024-05-16:
No dont... Thats the wrong one, that was for a different issue. Just stick with the one supplied in the image.

If youre display isnt working with that, its broken. Simple.

iandouglas, 2024-05-16:
I put your 2.8.9 epdconfig.py into the path of the waveshare examples, and still get the "ePaper busy" result on the 2.13" display and the 2.7 displays. (incidentally, changing my config.toml for pwnagotchi also does not display anything on these screens either)

While not impossible, I think it improbable that two brand new 2.7" displays out of the box (from two different vendors) are also broken on delivery.

Can you think of any other test I could try on the Pi itself to verify that the SPI/I2C bus isn't broken, or a pin didn't get soldered correctly, or something? This Pi worked until about a week ago... pwnagotchi still runs, I just have no display on it.

jayofelony, 2024-05-16:
The only thing I can think of is the connection to the screen is broken somehow

iandouglas, 2024-05-17:
I'll see if I can find one of my older Pi3 or Pi4 units just to confirm if the display works there ... I have one of those RPi keyboards somewhere but dunno if the screen will fit onto the pins on the back or not but worth trying.

iandouglas, 2024-05-18:
@jayofelony it's the Pi that was bad ... dunno how or why ... the 2.13" and 2.7" displays work great on a totally different Pi. Closing this issue since it's not a problem with pwnagotchi. Thanks for the pointers and help along the way!!!

The old Pi will be made into a headless unit.

closed 2024-05-18 by iandouglas