jayofelony/pwnagotchi #157: Bluetooth gets stuck after losing connection

matrix224, 2024-05-19:
- Pi Zero 2w
- Pwnagotchi 2.8.9

I have an odd issue with Bluetooth using this repo that I did not experience when using Aluminum-Ice's repo. I'm not sure at this point what the underlying problem is, but I believe it is something software related rather than hardware. I haven't seen anyone else complain about this, so I'm also curious if I'm really the only person in existence who has had this issue or what.

It basically boils down to:
- Pwnagotchi connects to my phone (Android) fine
- I walk away / something happens so that the bluetooth gets cut off from the device
- It cannot reconnect at all without an entire device restart
    -   Restarting the bluetooth service does not fix it

Does not matter if it's auto/AI or manual, it's the same deal.

Once the device has disconnected, BT-Tether starts printing out `[ERROR] org.bluez.Error.InProgress: Operation already in progress` whenever it tries to start its connection procedure. This error specifically is thrown when the plugin calls `StartDiscovery` on the BT device.

Using `dmesg | grep Bluetooth`, the latest things I see are:
```
[  366.883864] Bluetooth: hci0: link tx timeout
[  366.883888] Bluetooth: hci0: killing stalled connection <redacted>
[  367.908292] Bluetooth: hci0: link tx timeout
[  367.908330] Bluetooth: hci0: killing stalled connection <redacted>
[  368.899859] Bluetooth: hci0: command 0x0406 tx timeout
[  371.267785] Bluetooth: hci0: command 0x2005 tx timeout
[  371.267785] Bluetooth: hci0: Opcode 0x2005 failed: -110
[  373.283784] Bluetooth: hci0: command 0x200b tx timeout
[  373.283784] Bluetooth: hci0: Opcode 0x200b failed: -110
[  431.491819] Bluetooth: hci0: command 0x2005 tx timeout
[  431.491876] Bluetooth: hci0: Opcode 0x2005 failed: -110
```
The stalled connection is when I turned my phone off, and the following messages about command timeouts and opcode failures are when the plugin is trying to call `StartDiscovery` and subsequently `StopDiscovery`.

I have tried:
- Modifying the plugin to ignore the `InProgress` exception and continue processing
   - My assumption was that maybe it was already scanning and calling it again was causing an issue, but this did not solve it. Once it got to `StopDiscovery` it just threw some other error anyway and then continued to fail 
- Cycling the power using power on/off (both manually and programmatically through the plugin)
    - This just ends up leaving the device off, since trying to turn it back on results in `org.bluez.Error.Failed: Authentication Failed` 
- Restarting the service (both manually and by modifying the plugin to programmatically do it) 
    -  This gives the same thing as above
- Updating the kernel to 6.6.28
- Updating Bluez to 5.76 (fully removed bluez, blues-tools, etc first)
- Both of the above combined
- Switched devices
    - My first pwnagotchi was using Aluminum-Ice and my second one is using this repo on a second raspberry pi. I figured at this point maybe it's a hardware issue, so I tried switching the SD card to the first pwnagotchi's pi, but still no luck
 - Full reinstall several times

I'm assuming there might be some other package that's messing with the device somehow, but I'm really just not sure what else to do at this point.

letsgo245, 2024-05-20:
thats ehats been happening to my pwnagotchi altogether, our walk away from the computer and then it will disconnect from the Wi-Fi.

matrix224, 2024-05-21:
I believe I've found the source of this: nexmon.
I get the gist of what nexmon is and does, but I won't pretend I fully understand its usage or how it works, because I don't. 
I'm not condoning anyone to do what I did below since I don't fully know the ramifications; rather I just want the right people who understand all of this more than myself to see it so that they can make the appropriate moves or advisories as to what should be done.

After looking around some more (because I'm too hyperfixated on making things work I guess), I noted that the nexmon firmware patches that get installed in the build script had names like `BCM43430A1`. I recalled seeing something like that near the start of me doing `dmesg | grep Bluetooth` earlier.
So, I went back in and did it again to see what was there, and sure enough I see this:
```
[    9.546477] Bluetooth: hci0: BCM: chip id 94
[    9.546910] Bluetooth: hci0: BCM: features 0x2e
[    9.548150] Bluetooth: hci0: BCM43430A1
[    9.548165] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[    9.561112] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.raspberrypi,model-zero-2-w.hcd' Patch
[   10.329098] Bluetooth: hci0: BCM: features 0x2e
[   10.334388] Bluetooth: hci0: BCM43436 37.4MHz Class 1.5 RaspBerry Pi Zero2 [Version: 1017.1042]
[   10.334451] Bluetooth: hci0: BCM43430A1 (001.002.009) build 1042
```
Okay, so clearly I was right that nexmon is doing something here.
Looking at the install scripts, I see it's putting these firmware patches into `usr/lib/firmware/brcm'. So I go there, and sure enough I see:
```
BCM43430A1.hcd
BCM43430A1.raspberrypi,3-model-b.hcd
BCM43430A1.raspberrypi,model-zero-2-w.hcd
BCM43430A1.raspberrypi,model-zero-w.hcd
```
I know the nexmon patches are required for allowing the pwnagotchi to do its actual job, but at this point, I'm just running on a spare SD card I have around for science. So fuck it, what happens if I just rename the `BCM43430A1.raspberrypi,model-zero-2-w.hcd` patch to end in `.bak` and reboot?

The results:
```
[    9.864842] Bluetooth: hci0: BCM: chip id 94
[    9.865527] Bluetooth: hci0: BCM: features 0x2e
[    9.867048] Bluetooth: hci0: BCM43430A1
[    9.867096] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[    9.883975] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.hcd' Patch
[   10.552761] Bluetooth: hci0: BCM: features 0x2e
[   10.554068] Bluetooth: hci0: BCM43438A1 37.4MHz Raspberry Pi 3-0141
[   10.554090] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0508
```
So it seems to still be using the patch, but just not the Pi Zero 2w model one; instead it looks like just `BCM43430A1.hcd`, which is one for the Pi 3 (?)
In any case, for science, I continue my test and see what happens. I get my bluetooth going (I had to set it up again by re-pairing + trusting it), and it connects with no problem. 
I turn my phone off, and leave it off for 5+ minutes for good measure. I check the pwnagotchi logs, and to my shock, I see no errors from BT-Tether whatsoever. I turn my phone back on, and wouldn't you know it, it reconnects just fine.
I repeated this several times and every time, it appeared to work just as I would've hoped: it reconnected with no issue!

So naturally, I then went to look at my Aluminum-Ice build to see what was going on there in terms of these firmware patches, considering this seemed like the likely target.

In the `usr/lib/firmware/brcm' folder, I only see the `BCM43430A1.hcd` file. The other ones I had here that were for specific models were not present. When I run `dmesg | grep Bluetooth`, I also don't see any of the lines I posted about in regards to the BCM patches. So I'm not sure if this is due to some difference in logging between Aluminum-Ice's older bluez version (5.50) and this one's newer one (5.66), or if the nexmon used in Aluminum-Ice's patch is just not doing anything with bluetooth. 

In any case, I put this change on my main pwnagotchi and let him do do his thing after all of this was said and done, and he has been happily going about his business for about an hour now with no noticeable issues. The bluetooth reconnects fine after I walk off and come back, and hes been eating all the WiFi around here like the hungry bastard he is.

jayofelony, 2024-05-21:
Long answer in short;

I have installed all nexmon firmware patches for all devices and aluminum has not (read just 1). The 2w uses the bcm43430 wifi chip firnware as of raspios bookworm. In bullseye however it was using bcm43430 and/or bcm43436.

matrix224, 2024-05-22:
> I have installed all nexmon firmware patches for all devices and aluminum has not (read just 1). The 2w uses the bcm43430 wifi chip firnware as of raspios bookworm. In bullseye however it was using bcm43430 and/or bcm43436.

Got it, thank you for clarifying

> And nexmon patch does nothing with bluetooth.

Yeah I didn't think it did, but it clearly showed up in the logging for the bluetooth device. That's really the only reason I went down this path. I wish I had a better explanation for what it's doing or why taking `BCM43430A1.raspberrypi,model-zero-2-w.hcd` away and (somehow) making it use `BCM43430A1.hcd` made it work, but I don't.

mrbonkerz, 2024-06-02:
I'm experiencing the same issue. Has the workaround been effective for you?

I tried the same approach but couldn't reconnect when the bluetooth disconnects. Did you only rename the file, or did you also need to downgrade bluez to version 5.5?

matrix224, 2024-06-02:
For my final testing of it to see what all is needed, I reinstalled the image so everything was completely fresh, and then just renamed the file. I didn't touch anything else (excluding the obvious things like setting up config.toml, pairing bluetooth). It worked fine for me after doing just that.
From my over all testing, if you did this on an already running / configured setup, you'd have to re-pair your bluetooth again for it to connect properly after renaming the file. So if you didn't do that, then I'd at least give that a shot to see.

jayofelony, 2024-06-03:
I will add this to the build, renaming the filename.

closed 2024-06-03 by jayofelony

mrbonkerz, 2024-06-03:
> For my final testing of it to see what all is needed, I reinstalled the image so everything was completely fresh, and then just renamed the file. I didn't touch anything else (excluding the obvious things like setting up config.toml, pairing bluetooth). It worked fine for me after doing just that.
> 
> From my over all testing, if you did this on an already running / configured setup, you'd have to re-pair your bluetooth again for it to connect properly after renaming the file. So if you didn't do that, then I'd at least give that a shot to see.
> 
> 
> 
> Edit:
> 
> As an aside just for the record, I did not have this problem at all on my recent Raspberry Pi 4B install. I've only used a 4B and a Zero 2W (two of them actually), so I don't know about any other hardware but of those two pieces of hardware, the Zero 2W is the only one that had this issue for me so far (on two different installs on two different Zero 2Ws).

Thanks! It worked well for about 12 hours. After that, I had to reset it again. It seems the Pi isn't releasing the connection when the phone is out of range.