jayofelony/pwnagotchi #158: [BUG] fixing "GPIO busy"

iandouglas, 2024-05-20:
**Describe the bug**
The gpio_buttons plugin fails with some hardware because of the kernel changes with GPIO recently.

**To Reproduce**
Steps to reproduce the behavior:
1. ...
2. ...
3. ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
I mentioned it in the discord community here: 
https://discord.com/channels/717817147853766687/717818155061346427/1241236096659099669

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.8.9
 - Type of hardware: RPi Zero 2W, Waveshare 2.7inch e-Paper with 4 buttons

**Additional context**

```bash
sudo apt remove python3-rpi.gpio
sudo apt update
sudo apt install python3-rpi-lgpio
```

No other code changes were needed, and I can use the displays 4 buttons as GPIO 5, 6, 13 and 19 with the gpio-buttons plugin. This is purely a drop-in replacement package. Both `rpi.gpio` and `rpi-lgpio` import as the same module name, so while you can probably have both installed, it'll confuse your environment as to which module code to use.

letsgo245, 2024-05-21:
nice

jayofelony, 2024-05-29:
This is a nice update! I am testing this a we speak with a waveshare v2

jayofelony, 2024-06-03:
Working and implemented.

closed 2024-06-03 by jayofelony