jayofelony/pwnagotchi #161: Allow us to more easily cause agent.recon or something similar

iandouglas, 2024-05-20:
**Describe the solution you'd like**

I'm unsure whether my pwnagotchi is scanning all AP's and all channels ALL the time or if it's "sleep" mode is maybe to allow the unit to cool down or give more CPU time to the AI or something.

But when I'm walking around, I'd love to program a GPIO button that immediately triggers a "hey, pause anything else, or wake out of 'sleep'/'waiting' mode and immediately start aggressively looking for APs and scanning channels.

The only way I've found to do this so far is to use the gpio_buttons plugin to run a script that forces a restart of pwnagotchi in 'auto' mode, even if it was *already* in auto mode, but now I have to wait a while for it to cleanly shut everything down, start back up again, and then eventually it goes into monitoring mode.

For people without a GPIO button, having a web UI button for bluetooth/usb tethering would be a nice addition as well.

Ideally this wouldn't need a full restart of pwnagotchi. If the AI is already doing its thing, obviously let it finish and flush to disk, but if the code is in sleep/wait mode, it'd be nice to immediately go into "look around" mode.

rai68, 2024-05-26:
Sleeps are used for waiting for reply packets after sending out packets, It doesnt normally sleep because its bored or anything. Or stop attacking unless its blinded

iandouglas, 2024-05-26:
@rai68 so it's always scanning? The UI only shows that it's "looking around" on a periodic basis. If it's always scanning no matter what, and trying deauths, etc, regardless of the UI, then this can be closed.

rai68, 2024-05-29:
the webui is correct and "looking around" means its looking at what access points and devices are new, and ones that are no longer near. But there's no way to speed it up because of the waiting that needs to be done for devices to respond to attack packets. If we just keep scanning you never have any time to listen for a response.

closed 2024-06-17 by rai68