jayofelony/pwnagotchi #166: is there a way to share brain.nn across multiple adapters?

iandouglas, 2024-05-25:
**Is your feature request related to a problem? Please describe.**
It's not so bad that changing /boot/firmware/config.txt to disable wifi when you want to use an external wifi adapter or not, but it'd be nice if pwnagotchi could 'share' the AI brain across adapters.

**Describe the solution you'd like**
I'd love a technical explanation as to why the AI needs a different brain every time we change adapters. Sometimes I want to be more "discrete" with my pwny so I'll switch my config.txt, remove my external adapter, and reboot, but now the AI won't do anything.

**Describe alternatives you've considered**
I've debated writing some shell scripts that auto-switch the brain.nn / brain.json files via symlinks to be something like brain-internal.nn and brain-external.nn but it'd be nice to have the same stats accumulate altogether regardless of which adapter I'm using.

**Additional context**
If I have all of my pwny's dump pcap files to a central hub, and all pwny's sync all of those pcap files onto their own filesystems while charging at home, will the AI learn about all of them on each device?

rai68, 2024-05-26:
brain requires the exact same channels
so 5ghz to 2.4ghz wont work
and 2.4ghz to 5ghz wont work
but 5ghz to 5ghz (adapters must share same channels, **no more no less**) is fine, same with 2.4ghz

Basically, imagine growing up your entire life with 2 arms, but then randomly gaining an arm, you would have no idea how to use that arm, because you never learnt how to move it, because it just appeared one day. But sadly the AI cant learn like us and figure out to use more than 2 arms

iandouglas, 2024-05-26:
@rai68 Thanks for the explanation.

If 2.4GHz has 14 channels, 5GHz has 45 channels, 6GHz has 60 channels, then why not just build the brain to account for up to, say, 200 channels, and set up the known channel numbers? Then it won't matter which adapter you're using.

rai68, 2024-05-29:
because the AI does not know how to deal with less channels either, its like trying to use an arm you had that no longer exists, it would get confused and the AI would just crash. And not all adapters use all the available channels, and not all countries support all channels on the standard. the AI must be create with the channels it has access too and must always have those channels available

rai68, 2024-05-29:
Also no the captured files are useless to the AI, it only learns on each epoch with the information below

1. amount of APs on each channel in its brain
2. amount of devices on each channel connected to which AP
3. theres 1 more i think but i have forgotten it.

With those you can extract the best channels to spent time on

closed 2024-06-17 by rai68