jayofelony/pwnagotchi #173: OTG Connection Issues

xcs011, 2024-06-21:
Attempting to setup my new pwnagotchi on a brand new raspberry pi zero 2 W and I have been having consistent issues with OTG functionality.

The configs appear to be setup correctly, and my computer is able to recognize that there is an rdnis device connected via ethernet. 
However, I am still completely unable to ssh or ping the device, all of the above return timeout errors.

I have attempted using an older raspberry pi zero wh as well to rule out an error with the onboard USB controller, same error.

Do you have any ideas how to fix the functionality?

xcs011, 2024-06-21:
I have tried multiple cables as well, although I only have one OTG adapter, so that could be the issue. 
I will be getting another in Saturday as well as a USB-A to USB-A cable.

Currently I have been using a USB-C to USB-A cable since I did not have A to A, it could possibly be a USB-C issue?

In the meantime while I wait for parts, any support is appreciated.

wpa-2, 2024-06-21:
have you followed this 
https://pwnagotchi.org/getting-started/first-run-windows/index.html#ssh-into-your-pi

xcs011, 2024-06-21:
> have you followed this https://pwnagotchi.org/getting-started/first-run-windows/index.html#ssh-into-your-pi
Oops I skipped the configure section.
It now responds to pings, but responds with permission denied when attempting to ssh into <my username>@10.0.0.2

xcs011, 2024-06-21:
Nevermind, that was my own stupidity.
Yeah it works fine now, thank you.

closed 2024-06-21 by xcs011