jayofelony/pwnagotchi #174: Pwangotchi says connected when my phone is not

MimoAW, 2024-06-21:
I have an Google pixel 7 Pro I always used to be able to view the web UI through Bluetooth on my phone but one day it just stopped working the pwangotchi says connected but my phone is not connected. I have tried unpairing and untrusting and then repairing and trusting but it's still not working. It will also sometimes say be but then it'll just go back to connected.
Could I please have some help

wpa-2, 2024-06-21:
Follow this 
https://pwnagotchi.org/common-issues/index.html#instructions

MimoAW, 2024-06-21:
I have followed that guide previously. It did not solve my issue

wpa-2, 2024-06-21:
> I have followed that guide previously. It did not solve my issue

The guide works perfectly so there's a good chance you have messed something up, have you made sure tethering is enabled.

MimoAW, 2024-06-21:
Yes, Bluetooth tethering is enabled in hotspot. It was working perfectly fine but the next time I powered it on no changes to my phone or pwangotchi he just wouldn't connect and I did follow the guide multiple times

MimoAW, 2024-06-21:
I have fixed it. I stupidly had Android phone Bluetooth tether set to false 🤦 thanks for your help

closed 2024-06-21 by MimoAW