jayofelony/pwnagotchi #175: Problem on the display

coutud, 2024-06-25:
Hello i have a problem on the display The battery level that is displayed overlaps with the bt connection indication how i can separe them

Thanks

jayofelony, 2024-06-25:
Either use tweakview plugin or edit the corresponding plugin you wish to edit the location of.

coutud, 2024-06-25:
HelloOk but i don’t know how to do itThanks Envoyé de mon iPhone

renewedbullet, 2024-06-26:
https://pwnagotchi.org/customization/tweak-view/index.html

Tweak view is a plugin for pwnagotchi that allows you to move UI elements.

rai68, 2024-08-13:
> https://pwnagotchi.org/customization/tweak-view/index.html
> 
> Tweak view is a plugin for pwnagotchi that allows you to move UI elements.

Closing, fix is above

closed 2024-08-13 by rai68