jayofelony/pwnagotchi #177: Display Preset Question

xcs011, 2024-06-27:
Does the "pitft" option refer to the line of adafruit models? (For instance [this](https://www.adafruit.com/product/2298))
Would I need special configuration options in the config for a TFT instead of E-Ink?

I have not been having much work actually getting it to work.

Pwnagotchi is running on a Raspberry Pi 3b+ and confirmed to work with an E-ink display.

xcs011, 2024-06-27:
I found a guide that allegedy can be followed to enable the pitft as well 
https://github.com/evilsocket/pwnagotchi/issues/124

I will keep the issue updated

xcs011, 2024-06-28:
It did not work, the info appears to be outdated.

xcs011, 2024-06-28:
To use the pitft with this pwnagotchi fork, should it be setup in frame buffered mode?
Is there anything I need to add to the config.toml?

rai68, 2024-07-29:
have you tried `ui.display.type = 'pitft'` in config?

rai68, 2024-08-13:
Closing due to inactivity. Open a new issue if it persists

closed 2024-08-13 by rai68