jayofelony/pwnagotchi #181: [BUG]

1ch0712, 2024-07-06:
**Describe the bug**
Don`t work internet

Firmware: 2.8.9
I use windows to give out internet but it doesn't show up. I try pinging but no response comes back. Used a script "win_connection_share.ps1" but it didn't produce any results.
Before that, I was using version 1.5.5 and there I was able to give away internet.

jayofelony, 2024-07-06:
Follow the wiki or https://pwnagotchi.org .

Closing this as it is not a bug, internet sharing works how it should.

closed 2024-07-06 by jayofelony

1ch0712, 2024-07-07:
Correct me if I'm wrong but neither the site nor the wiki has any information about connecting gotchi to the internet or solving the problem with it. Again, I connected to the internet image 1.5.5 but on 2.8.9 I am not succeeding.