jayofelony/pwnagotchi #183: [BUG] waveshare 4 not working

Raeloen, 2024-07-09:
**Describe the bug**
Hi there, i just installed the latest Image but i can't get a waveshare v4 Display to work. I tried a solution which worked on 1.5.5 but no luck.
It came to my attention that the waveshare 4 screen has already been implemented so i suggest it's a bug.

I am using a raspberry pi zero wh.

jayofelony, 2024-07-10:
Its not a bug, its your display or a misconfiguration. I will need more info.

Try using --wizard.

jayofelony, 2024-07-10:
It cannot be the image, you would be the only one having an issue. Out of 10k devices made so far.

closed 2024-07-10 by jayofelony

wpa-2, 2024-07-10:
> I actually did and used waveshare_3 but it did not work. The display worked on 1.5.5 and another Image so it can't be the display

I use the image on 2 different pizeros with the v4 screen so it has to be something your end.

renewedbullet, 2024-07-10:
> I actually did and used waveshare_3 but it did not work. The display worked on 1.5.5 and another Image so it can't be the display

When did you have it working last on 1.5.5? Are you part of the discord, did you solder your own pins or pre installed pins? A picture of how its assembled could help with troubleshooting, but currently there is probably a solid 2-3k users using the v4 screen if not more and 0 reported issues outside of user error, primarily soldering an or damaged display cables.

Raeloen, 2024-07-10:
Hi there! It worked yesterday but reddit told me to forfeit the pwnagotchi.ai Image and check pwnagotchi.org.
There i found this Image and installed it. Are there any discords who can help a total rookie out?

renewedbullet, 2024-07-10:
Yes the discord is linked on the org website.