jayofelony/pwnagotchi #185: [BUG]

jonpotz, 2024-07-22:
**Describe the bug**
For waveshare 3.7 inch screen on a raspberri pi4 the screen was not oriented correctly. It was small, didn't fill up the full screen even when using the rotate function. If using the rotate function only the status bar showed on the screen. Please see attached pictures. Latest Jayofelony build.

**To Reproduce**
Steps to reproduce the behavior:
1. Use display type waveshare3in7 on an rpi4.
2. Boot and the screen will show up funky.

**Expected behavior**
A full screen that has the correct resolution.

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version ( latest build ) 
 - pi4 - pisugar3 - gps-module - panda wifi adapter

**Additional context**

The solution was to:

Navigate to: /usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw

You will find a file waveshare3in7.py

Go down to class Waveshare3in7(DisplayImpl): and under def layout(self):

You will find that the self._layout['width'] and self._layout['height'] are swapped.

Change it to: self._layout['width'] = 480 self._layout['height'] = 280

The screen will now be orientated the correct way.

I also had to play around with the x y axis of the different labels, here is my version:

def layout(self):
    fonts.setup(20, 19, 20, 45, 35, 19)
    self._layout['width'] = 480
    self._layout['height'] = 280
    self._layout['face'] = (0,34)
    self._layout['name'] = (35,105)
    self._layout['channel'] = (0, 0)
    self._layout['aps'] = (75,0)
    self._layout['uptime'] = (377,0)
    self._layout['line1'] = [0, 25, 480, 25]
    self._layout['line2'] = [0, 255, 480, 255]
    self._layout['friend_face'] = (0, 146)
    self._layout['friend_name'] = (40, 146)
    self._layout['shakes'] = (0, 258)
    self._layout['mode'] = (430, 258)
    self._layout['status'] = {
        'pos': (225, 35),
        'font': fonts.status_font(fonts.Medium),
        'max': 21
    }
    return self._layout

I would also installed Tweak View plugin to fine tune where you want the labels.

Anyways, I hope this helps someone. Maybe if Jayofelony sees this he can update the code.

jayofelony, 2024-07-23:
I will, thank you.

rai68, 2024-08-13:
fixed with 48b3d6dcb5125f5b23ec04dc8dae9ed205a48d3a