jayofelony/pwnagotchi #187: When AI is disabled, the channel list never changes so pwnagotchi cannot discover new nets

Sniffleupagus, 2024-07-31:
**Is your feature request related to a problem? Please describe.**
It is frustrating that AI uses so much CPU for negligible or undetectable utility. It is effectively a channel and timing randomizer. It is unclear that it develops emergent behavior.  It just eats up CPU.

If AI is disabled, the channel list is static. Pwnagotchi cannot discover new nets, unless they are on the channels listed in personality.channels in config.toml.

**Describe the solution you'd like**
A solution would be to change AUTO mode to alter the channel list.

**Describe alternatives you've considered**
I have tried just running it and scanning all available channels (personality.channels = []).  It works, but since most channels are empty most of the time (it seems), pwny wastes a lot of time on unused channels, instead of pwning active APs.
**Additional context**
Add any other context or screenshots about the feature request here.

itsOwen, 2024-07-31:
Can we use a scoring system instead of an "AI" i think it would be way better as using AI makes my rpi heat way too much.

What we can do is and its a rough idea:

We'll add a new part to that keeps track of how busy each WiFi channel is.

Every time the Pwnagotchi looks at a channel, it'll count how many WiFi networks and devices are there.

Channels with more stuff get a higher score.

When it's time to search for networks, the Pwnagotchi will usually pick one of the busy channels, But it will also pick the channel with low in activity afterwards so its like priority thing.

We can change how often it switches channels by tweaking some settings probably.

Sniffleupagus, 2024-07-31:
I created this issue to comply with submitting a PR. I did something like that, but not keeping track of scores.  Any channels that pwnagotchi finds an AP on will be part of the next scan, plus a random selection of the remaining channels. Empty channels only get scanned every few epochs, and don’t waste as much time scanning nothing. But if you’re sitting somewhere for 10 minutes, every channel will have been scanned many times

Sniffleupagus, 2024-07-31:
PR of above implementation https://github.com/jayofelony/pwnagotchi/pull/188

closed 2024-11-24 by jayofelony