jayofelony/pwnagotchi #189: Can't get WaveShare V4 Working

djvdberg, 2024-08-06:
Hallo,

Running latest image on Raspberry Pi zero 2W with waveshare V4

Config set to Waveshare_4

No display, anything else I should do?

Display - https://www.pishop.co.za/store/213inch-e-ink-display-hat-for-raspberry-pi-250x122
Got a V4 sticker on the bottom

Config

ui.display.enabled = true
ui.display.type = "ws_4"
ui.display.rotation = 180

Thanks!

ianfs2, 2024-08-07:
ui.display.type = "waveshare_4"

djvdberg, 2024-08-07:
Updated but still no display?

ui.display.enabled = true
ui.display.type = "waveshare_4"
ui.display.rotation = 180

ianfs2, 2024-08-07:
In which file are you configuring?

translated by google.

djvdberg, 2024-08-07:
sudo nano /etc/pwnagotchi/config.toml

ianfs2, 2024-08-07:
It should work...
I use the same settings and have never had this problem.

Raspberry Pi zero 2W & waveshare V4

ui.display.enabled = true
ui.display.rotation = 180
ui.display.type = "waveshare_4"
ui.display.color = "black"

jayofelony, 2024-08-07:
run:

`sudo systemctl stop pwnagotchi && sudo pwnagotchi --debug`

See if any errors pop up

djvdberg, 2024-08-07:
Getting a bunch of these

`[2024-08-06 20:24:34,802] [INFO] initializing waveshare v2in13_V4 display
[2024-08-06 20:24:34,802] [DEBUG] Got here
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-08-06 20:24:35,100] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-08-06 20:24:35,101] [DEBUG] e-Paper busy release
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-08-06 20:24:35,102] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-08-06 20:24:35,103] [DEBUG] e-Paper busy release
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-08-06 20:24:35,108] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-08-06 20:24:35,109] [DEBUG] e-Paper busy release
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-08-06 20:24:35,120] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-08-06 20:24:35,121] [DEBUG] e-Paper busy release
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-08-06 20:24:35,951] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-08-06 20:24:35,952] [DEBUG] e-Paper busy release
`

djvdberg, 2024-08-07:
Also just noticed now

**default.toml :**
ui.display.enabled = false
ui.display.rotation = 180
ui.display.type = "waveshare_4"

**config.toml**
ui.display.enabled = true
ui.display.type = "waveshare_4"
ui.display.rotation = 180
ui.display.color = "black"

jayofelony, 2024-08-07:
How is your soldering, or did you use hammer pins?

djvdberg, 2024-08-07:
Think they pretty good, fluxed and all

Was thinking I'll try it on the SPI port tomorrow

Screen is brand new, but pi was used for other stuff, so not sure if something might be broken

[image]
[image]

jayofelony, 2024-08-08:
That looks good, if there are no noticable errors in --debug than it must be either faulty display or broken pi. Display being the one that breaks easier.

AleTornesello, 2024-08-09:
It seems to me that there are some pins that are cold soldered or with little flux.
[image]
Before buying new hardware I suggest you try to resolder all the pins at a slightly higher temperature and with plenty of flux.

code-for-kaffee, 2024-08-12:
ui.display.type = "waveshare_3"

Also as @AleTornesello suggested, try resolder

code-for-kaffee, 2024-08-13:
> i wouldnt use _3, _4 is best with a V4 sticker, if no display after reheating those pins display may be damaged

_3 and _4 have the same configs, its duplicated code.

AchimGrolimund, 2024-08-30:
hey @djvdberg  dose it work now for you?

i have a similar issue but with other Hardware (Raspberry pi 4B)
https://github.com/jayofelony/pwnagotchi/issues/200

I see also only e-ink is busy but nothing happens on the display.

closed 2024-11-24 by jayofelony