jayofelony/pwnagotchi #198: I2C Display

Xanndiane, 2024-08-28:
Hi, I would like to know how to use the ssd1306 i2c screen? It is in the repository files, but it is not in the image.

RasTacsko, 2024-09-21:
Hi
[I made a bash script](https://github.com/RasTacsko/Pwnagotchi-plugins/blob/main/displaydriverupdate.sh) that downloads my pwny repo with the updated display drivers, and copies it to the necessary folders, so if someone afraid to do it manually, it can be automated... Be aware that is made with chat gpt, and it is need internet connection of course, so use at your own risk, and modify as necessary...
Just create an .sh file in /home/pi or copy it with sftp and run it after chmod +x all feedbacks are welcome of course

Xanndiane, 2024-09-27:
> Hi [I made a bash script](https://github.com/RasTacsko/Pwnagotchi-plugins/blob/main/displaydriverupdate.sh) that downloads my pwny repo with the updated display drivers, and copies it to the [...]

Unfortunately it doesn't work for me, I already wrote to you about it on Reddit, have you tested it yourself?

RasTacsko, 2024-09-27:
Hi
I haven't checked up on reddit, but modified the scripta bit, there was a target directory error.
jays 2.8.9 release is using python 3.9 for the 32bit image, now the script will ask for what operating system you have and should update the necessary files and folders.
Try it again and feedback please, so I can fix the issue.

Xanndiane, 2024-09-27:
I tested the screen outside Pwnagotchi and it works. Maybe there is some way to get the state and other data from pwnagotchi to output them through third-party code? (Not a ready-made bitmap image, but data in the form of text)

Xanndiane, 2024-09-27:
Okay, I'll try in a few hours.

Xanndiane, 2024-09-27:
Спасибо

Xanndiane, 2024-09-27:
> Hi I haven't checked up on reddit, but modified the scripta bit, there was a target directory error. jays 2.8.9 release is using python 3.9 for the 32bit image, now the script will ask for what [...]

Now its work. Thanks bro.
[image]

closed 2024-09-27 by Xanndiane