jayofelony/pwnagotchi #200: [BUG] waveshare v4 + Raspberry 4B busy error

AchimGrolimund, 2024-08-29:
**Describe the bug**
I can't get the Waveshare v4 to work. 
bought it:https://www.digitec.ch/de/s1/product/waveshare-250x122-213inch-schwarz-weiss-e-ink-display-raspberry-pi-hat-entwicklungsboard-kit-24963494?supplier=8244233

some logs on normal mode:
```
/usr/local/lib/python3.11/dist-packages/stable_baselines3/common/vec_env/base_vec_env.py:243: UserWarning: You tried to call render() but no `render_mode` was passed to the env constructor.
  warnings.warn("You tried to call render() but no `render_mode` was passed to the env constructor.")
```
logs in debug mode:
```
[2024-03-15 15:49:37,381] [INFO] Session-stats plugin loaded.
WARNING:root:ui.fps is 0, the display will only update for major changes
DEBUG:root:Got here
[2024-03-15 15:49:37,391] [WARNING] ui.fps is 0, the display will only update for major changes
INFO:root:initializing waveshare v2in13_V4 display
[2024-03-15 15:49:37,391] [DEBUG] Got here
[2024-03-15 15:49:37,392] [INFO] initializing waveshare v2in13_V4 display
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-03-15 15:49:37,516] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-03-15 15:49:37,517] [DEBUG] e-Paper busy release
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-03-15 15:49:37,517] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-03-15 15:49:37,518] [DEBUG] e-Paper busy release
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-03-15 15:49:37,520] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-03-15 15:49:37,520] [DEBUG] e-Paper busy release
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-03-15 15:49:37,529] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-03-15 15:49:37,529] [DEBUG] e-Paper busy release
INFO:root:[AI] loading /root/brain.json
[2024-03-15 15:49:38,098] [INFO] [AI] loading /root/brain.json
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-03-15 15:49:38,110] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4:e-Paper busy release
[2024-03-15 15:49:38,111] [DEBUG] e-Paper busy release

``` 

**Environment (please complete the following information):**
 - Pwnagotchi version `2.8.9`
 - Image: `https://github.com/jayofelony/pwnagotchi/releases/tag/v2.8.9`
 - OS version `Linux paw 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux`
 - Type of hardware `Raspberry Pi 4B`
 - Any additional hardware used `waveshare v4 Rev2.1` connect with pins (not soldered)

rai68, 2024-08-30:
What does connect with pins (not soldered) mean, can you show how its connected

AchimGrolimund, 2024-08-30:
Hey @rai68 here the picture.

[image]

AchimGrolimund, 2024-08-30:
i will first try to find a example script for my ESP32 to see if really the HW is the problem and then i will contact them.

:-( First e-Paper and broken or broken after 1 day :-( bad vipes :D

rai68, 2024-08-30:
Yes forgot about the example scripts, theres a python one you can download for the Pis

AchimGrolimund, 2024-08-30:
do you know just one for the Rasperys? i have only found something for the PICO

rai68, 2024-08-30:
https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT_Manual#Python_2

```
git clone https://github.com/waveshare/e-Paper.git
cd e-Paper/RaspberryPi_JetsonNano/
Download the demo (you can skip this step if you have downloaded it.).
wget https://files.waveshare.com/upload/7/71/E-Paper_code.zip
unzip E-Paper_code.zip -d e-Paper
cd e-Paper/RaspberryPi_JetsonNano/
Alternate decompression method:
sudo apt-get install p7zip-full
7z x E-Paper_code.zip -O./e-Paper
cd e-Paper/RaspberryPi_JetsonNano/
Run the demo:
# Make sure it's in e-Paper/RaspberryPi_JetsonNano/
cd python/examples/
python3 epd_2in13_V4_test.py
```

AchimGrolimund, 2024-08-30:
unfortunately, as you said, the hardware seems to be defective.
I have connected the display with the individual pins and the connector, as well as packed directly onto the board. but unfortunately both are unsuccessful. also always with the bussy log

i have now reported it to the seller and ordered 2 new ones ;-)

Thanks alot for the Help! that is awsome fast ;-)

closed 2024-08-30 by AchimGrolimund