jayofelony/pwnagotchi #201: [Question] Connect to Home WIFI instead Ethernet

AchimGrolimund, 2024-08-30:
you can connect the Raspy (4) via Ethernet cable to read out the data (ssh) or for the WebService.

Is it also possible to set the WIFI settings when flashing? so that the Raspi(4) always establishes a WIFI connection with the home network and you can call up the Web UI?

Or does this interfere with the scanning process of the pwnagotchi?

Best regards

RasTacsko, 2024-08-30:
When the pwnagotchi boots up, the bettercap service is setting up the wifi to monitor mode, so it is not able to connect to any wifi networks as usual. There are workarounds if you are using a usb wifi dongle so you will have one for scanning one for regular use.

AchimGrolimund, 2024-08-30:
that's what i thought. thanks for the quick reply.

closed 2024-08-30 by AchimGrolimund