jayofelony/pwnagotchi #204: [BUG] plugins disappear on Home Screen but remain in config file enabled

ScottishShaun123, 2024-09-10:
**Describe the bug**

I don’t know what happened but every plugin has disappeared from the Home Screen and the plugins screen but weirdly I can see them in the web CFG page and they are enabled as they were before. I can still see all the default plugins that came with the build and they are all working as far as I’m aware. 
[image]

**To Reproduce**
Steps to reproduce the behavior:
Unknown 

**Expected behavior**
Custom plugins should remain active and visible on Home Screen and also plugins page. They are seen as activated in webcfg but they are not.

**Screenshots*

**Environment (please complete the following information):**
 - Pwnagotchi version jayofelenoy image 2.8.9
 - OS version 
 - Type of hardware raspberry pi zero W
 - Any additional hardware used
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

jayofelony, 2024-09-10:
Whats your custom-plugin folder set to in your config.toml, and does it match with the location of the plugin files?

ScottishShaun123, 2024-09-10:
You are the man 😅 I feel so dumb now. I’ve only went and input a git link in place of where the folder location should be when adding custom plugins. All working again. I truly appreciate all the work you do for this project. 🙏

> Whats your custom-plugin folder set to in your config.toml, and does it match with the location of the plugin files?

closed 2024-09-10 by ScottishShaun123