jayofelony/pwnagotchi #206: [BUG] Rpi pi zero 2 W dose not boot

AchimGrolimund, 2024-09-19:
**Describe the bug**
the Raspberry will not start.
i downloaded the 64bit image with version 2.8.9 and flashed it to an sd with the raspberry imager. when i plugged it in, the green led has been lit constantly without interruption for an hour.
i have tried both ports (usb pwr and data) but the same result.

is it possible that this pwnagotchi img is not compatible with the rpizero2w?

**To Reproduce**
Steps to reproduce the behavior:
1. flash 64bit 2.8.9 to sd
2. start it with rpi zero 2 w
3. green light goes on constantly without blinking or else

AchimGrolimund, 2024-09-19:
[image]

wpa-2, 2024-09-19:
There's no bug it works fine with the pizero2! 
I suggest a reflash and start here 
https://github.com/jayofelony/pwnagotchi/wiki

AchimGrolimund, 2024-09-19:
i wil flush it again. in the meantime i test out the default 64 bit raspian and it works.

in 15 min i can gife a feedback.

wpa-2, 2024-09-19:
> ![image](https://private-user-images.githubusercontent.com/22026079/369133406-a908e4c1-04fe-43d8-83a1-f58e661d3212.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3NzMyNDYsIm5iZiI6MTcyNjc3Mjk0NiwicGF0aCI6Ii8yMjAyNjA3OS8zNjkxMzM0MDYtYTkwOGU0YzEtMDRmZS00M2Q4LTgzYTEtZjU4ZTY2MWQzMjEyLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOVQxOTA5MDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNWRkYWYxZWJlYzUxZTYyOWU0ZmI5OGJjMjM0ODY3OWY5M2VlMTc0NzBmNmFlZGY2MTAwYWQ2ZWUxYWYzNmM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.CPjC9kz6UI1lecUK74nZ878DQFO4_tRuqdH3Ju_1MVM)

Is this connected to your pc?  
Because your on the wrong port if so, you need to use data for it to show up as a network device.

AchimGrolimund, 2024-09-19:
hey @wpa-2  same. i flash it with the exact settings written in the Wiki.
Exactly the same behavior. Green LED is lightning constantly.

wpa-2, 2024-09-19:
> hey @wpa-2 same. i flash it with the exact settings written in the Wiki. Exactly the same behavior. Green LED is lightning constantly.

~Are you still using the power port?

AchimGrolimund, 2024-09-19:
> Is this connected to your pc? Because your on the wrong port if so, you need to use data for it to show up as a network device.

as i wrote. i triet booth. not it is only conectet to the data port and to my notebok. (mac)

AchimGrolimund, 2024-09-19:
> > hey @wpa-2 same. i flash it with the exact settings written in the Wiki. Exactly the same behavior. Green LED is lightning constantly.
> 
> 
> 
> ~Are you still using the power port? 

[image]
😉😉

wpa-2, 2024-09-19:
Follow this 
https://pwnagotchi.org/getting-started/first-run-mac/index.html

AchimGrolimund, 2024-09-19:
i know all those sites =) i have an allready running rpi4b. but i want to run int on the zero2 ;-)

as you can see, also the RNDIS is not shown in the settings.
[image]

hmm im wondering if it is posible to use the SDcard from the RPI4B inside the Zero2? booth are 64Bit.

proxy-the-great, 2024-10-10:
This is probably an issue with your rpi 02W?

I just purchased one along with a waveshare ink hat v4 and I have the same problem, I also noticed that when I try using any other Pwnagotchi image my Raspberry Starts and shuts down after a few seconds....this is probably an issue with our boards :(

Images I tested on my board: 
- https://github.com/aluminum-ice/pwnagotchi
- https://github.com/DrSchottky/pwnagotchi

I'll buy a new one in a few days....

AchimGrolimund, 2024-10-10:
Nope all other images are running.

It is onli this pwnagotchi was not realy works

proxy-the-great, 2024-10-10:
Oh....looks like I have a broken board :(

AchimGrolimund, 2024-10-13:
so i found out my "issue/problem" my SD card is to slow...

i have ordered a new one and now it works.

- i test it out with all the 3 SD-Cards (Intenso microSDXC UHS-I Performance 64GB)
Dose not work at all 

- now i got a new one (Samsung EVO Plus, microSDXC, 64 GB, U1, UHS-I)
It Works

hypernab, 2024-10-19:
Aren't they both U1 and UHS-1 tho?
I'm having the same issue with my SanDisk ultra so maybe i should consider getting a Samsung Evo now.

AchimGrolimund, 2024-10-19:
I will check later on the card performance.
What i know is the bad card has a read performance of 90Mb and a write of min 10Mb 
I guess that's not enough to boot.

My samsung card has 100Mb write and 130Mb Read

On my rpi 4B i use also sandisk this has 60Mb write and 100Mb Read.

So Check on your card performance i guess you should have at least read and write over 50Mb.

hypernab, 2024-10-19:
Yeah well turns out the sandisk has a write speed of 10-20 MB/s so i guess ill need a better one.

hypernab, 2024-10-19:
Update: i used the same sd card with a 32 bit image for Rpi Zero W instead 2 and it works completely fine
Moral of the story: Rpi Zero 2 w image is broken?

AchimGrolimund, 2024-10-19:
Dont think so.
As I explained. With my intenso slow performance card it dose NOT work on the zero pi 2.
Using the same image on a faster card (90mb+ read and write)
It works perfectly fine.

hypernab, 2024-10-19:
Will try that too

closed 2024-11-24 by jayofelony