jayofelony/pwnagotchi #207: Request for a High-Performance Jayofelony Version of Bettercap for Pwnagotchi and some ai improvements and suggetions

Flash-ed, 2024-09-20:
Hello ,

I propose the development of a more efficient Jayofelony version of Pwnagotchi that integrates Bettercap. The current version is sluggish and lacks robust attack capabilities. Key enhancements should include optimized man-in-the-middle attacks, packet injection, and DDoS functionalities. Furthermore, integrating features to enable Pwnagotchi to hack Wi-Fi networks seamlessly and update automatically when connected to the internet would significantly improve its effectiveness.

I can assist with the coding if you're interested, and I am willing to sponsor the project if we can successfully execute this correctly together. Thank you for considering this proposal.

Sincerely,  
Flash
(Flash-Ed  and FlashedCode) on GitHub)
Ethical Hacker and Programmer  
Part of Anonymous

wpa-2, 2024-09-20:
Man in the middle attacks , ddos , are you kidding? 
Pwnagotchi is based on bettercap, what are you actually talking about. 

Get coding let's see what you can come up with?

jayofelony, 2024-09-21:
You made me giggle this saturday morning, have a good day sir.

closed 2024-09-21 by jayofelony