jayofelony/pwnagotchi #209: [BUG] peer detection not working with Pi Zero W - are advertisement beacons sent at all? How to spot them?

hb9eue, 2024-09-27:
I and a Work-Mate have both set-up a pwnagotchi on a Pi Zero W.

They harvest handshakes, but never see each other as peers, not even when left beside each other for many hours.

Some google research found other user observing the same issue, all using a Pi Zero W. So it looks specific to that device.

So I attempted to dig further into how peer detection works. I didn't get too far into it tho. But I understand that this is done via custom beacon packages sent containing peer information.

I also learned (please correct me if I am wrong) that monitor mode and packet injection on the raspberries rely on altered WiFi chip firmware as this is not possible with stock firmware and that there are many issues with WiFi firmware behaving differently or trying to correct injected packets instead of letting them through.

So I started monitor mode on my notebook next to my Pwnagotchi and captured radio packets to see if I can spot those customer beacons...

I don't see any beacons not from AP's around me.

So I started wondering. Are those special peer advertisement beacons being sent at all? What would they look like when captured with wireshark on another host running in monitor mode?

-Benoît-

Saru13, 2024-10-16:
I noticed this too.

dj1ch, 2024-10-30:
> They harvest handshakes, but never see each other as peers, not even when left beside each other for many hours.

Strange. I used to run this on my Pi Zero W (with Jay's 2.8.9) image and advertisments worked just fine. I switched over to using my Pi Zero 2 W so I stopped using it. Auto mode helps especially if you want it to be on a fixed set of channels.

For my config I had to enable peer advertisement with something that looked like this

```toml
personality.advertise = true
```

I also kept `pwngrid` enabled, along with its whitelist and all that. Not sure if that's relevant at all.

> So I attempted to dig further into how peer detection works. I didn't get too far into it tho. But I understand that this is done via custom beacon packets sent containing peer information.

In order to handle the packets (extracting their info, etc) I had to do a specific process that pretty much matched what this [pwngrid-related file](https://github.com/jayofelony/pwngrid/blob/master/wifi/pack.go) does, except in reverse. It does a lot of things that I find hard to explain, but it seems to be a normal beacon frame with injected info, along with being put through some weird sort of algorithm that "chunks" the payload by inserting a couple of "magic" numbers here and there per every 255 bytes or so. Undoing it is mostly simple as it was already being done by pwngrid [here](https://github.com/jayofelony/pwngrid/blob/master/wifi/unpack.go). I have made a similar implementation in embedded C/C++ [here](https://github.com/dj1ch/minigotchi-ESP32/blob/main/minigotchi-ESP32/pwnagotchi.cpp#L131).

> What would they look like when captured with wireshark on another host running in monitor mode?

As for finding them you'll just have to keep hopping wifi channels until you happen to be on the same one as a pwnagotchi. It's not exactly being advertised as an AP here like a beacon frame but some programs or other devices such as the ESP32 marauder can identify the advertisement as such. Here is an [example](https://github.com/viniciusbo/m5-palnagotchi/blob/master/pwngrid_beacon.hexdump) captured pwngrid packet I found online, in a hexdump. Hope this helps!

closed 2024-11-24 by jayofelony