jayofelony/pwnagotchi #212: [BUG] No ping or traceroute on v2.8.9 via iPhone

MCKLtech, 2024-10-03:
**Describe the bug**

pwnagotchi is fully tethered and connected to iPhone, with hotspot enabled, but when SSH'd into the pwnagotchi, it is unable to reach the internet. Ping to 8.8.8.8 etc does not work and traceroute results in "Temporary failure in name resolution" In reverse, I can ping the iPhone IP address and the routes appear to be set up.

I've previously set up pwnagotchi with other images and this has worked before, it seems specific to this image. Notably, with this image, I have to use the IP address in the iPhone safari to access the webUI, when I previously could have used pwnagotchi.local:8080

**To Reproduce**
Steps to reproduce the behavior:
1. Flash image
2. Set up Bluetooth and tether to iPhone
3. SSH to Pi and attempt to ping, curl etc.

**Expected behavior**
Previous installs of pwnagotchi have worked correctly and been able to access the internet via the same iPhone.

jayofelony, 2024-10-03:
If you want to use hostname, install avahi-daemon package.

jayofelony, 2024-10-03:
On the first problem,are you still connected to the pc through usb? Because that may cause routing errors for internet.

MCKLtech, 2024-10-03:
> On the first problem,are you still connected to the pc through usb? Because that may cause routing errors for internet.

Yes, I'm connected via USB on the data port to MacOS. This has worked in the past as I would SSH to the pi and install packages etc, the Pi therefore using the tether from the BT.

Interestingly, in AUTO mode, the Pi itself can use the internet via BT as it can upload to services.

jayofelony, 2024-10-04:
In manual mode, it is static.

MCKLtech, 2024-10-04:
> In manual mode, it is static.

Can I ask what you mean by static? If you mean not scanning, capturing, then yes, I know that. I'm just curious how in AUTO mode it seems to be able to reach the internet and upload, but when I SSH into it, and try curl, it can't.

jayofelony, 2024-10-04:
Are you in auto, and also connected to usb/pc?

MCKLtech, 2024-10-04:
I've tried two scenarios:

**Manual**: Connected to MacOS via data port and Bluetooth to iPhone with hotspot. When I SSH to the device, I cannot curl or ping the internet. Curl etc has worked previously with other images.

**Auto**: Connected to MacOS via data port and Bluetooth to iPhone with hotspot. When I SSH to the device, I cannot curl or ping the internet. However, plugins that need internet access, for example to upload hashes, do work and uploads are successful.

jayofelony, 2024-10-05:
Only thing I can think of is that it is to do with metrics somehow. The preferred way of routing. Try disabling BT, and to curl again. See if that changes anything.

closed 2024-11-24 by jayofelony