jayofelony/pwnagotchi #220: pi/raspberry does not work at console for login (USB keyboard adapter to get local console access)

xelerated, 2024-10-30:
I am using a keyboard to log in locally, pi/raspberry do not work. From what i can tell from the docs that should work for SSH, so shouldnt it work for console? 

**To Reproduce**
Use Micro to USB A female to connect keyboard.
Boot up Pi with FRESH latest Image
pi/raspberry does not give me a shell

**Expected behavior**
Unable to log in from the console

**Screenshots**
not needed, login will not work

**Environment (please complete the following information):**
 - Pwnagotchi version 2.8.9 (latest, 32bit)
 - OS version Whatever is on that 2.8.9 image
- Pi 0 W 2

Should be able to log in using this method.

wpa-2, 2024-10-30:
Pizero 2 uses 64bit image, try that and report back.

xelerated, 2024-10-30:
I was incorrect, its a Pi 0 v1.1

xelerated, 2024-10-30:
Sorry, I am a moron. My pi imager was using old config that set a user/pass from a prior project. :( Computers are hard

closed 2024-10-30 by xelerated