jayofelony/pwnagotchi #221: Update onlinehashcrack.py

g4h4, 2024-10-30:
Kindly update module onlinehashcrack.py
New API documentation : https://www.onlinehashcrack.com/how-developpers-api-documentation.php
and https://www.onlinehashcrack.com/how-developpers-public-api-documentation.php

wpa-2, 2024-10-30:
> Kindly update module onlinehashcrack.py (pwnagotchi/plugins/default/onlinehashcrack.py) New API documentation : https://www.onlinehashcrack.com/how-developpers-api-documentation.php and https://www.onlinehashcrack.com/how-developpers-public-api-documentation.php

You could always do that and create a PR, some how I don't think this is top priority currently.

closed 2024-11-24 by jayofelony

g4h4, 2024-11-25:
As far as i understand, new changes are : listing ongoing tasks and add 'wpax' hashes.