jayofelony/pwnagotchi #225: Waveshare V4 (tri-color) screen not working, can't SSH

markpavlenko, 2024-11-09:
Hi guys!
**Describe the bug**
So I installed Pwnagotchi 2.8.9 on my hardware and everything worked fine. But after i re-installed the pwnagotchi again because of a problem with ssh, the screen stopped working, refreshing or showing something, it just didn't reset. It stuck on the previous installation screen with pwny's face. I tried looking at the logs, re-install with other plugins, but it simply won't work. I got it to the service after a while and they said it works ok on their RPi 3. They gave me it with a fresh "Waveshare 2.13inch e-Paper" screen. After, I tried doing it again and reinstalling everything but it still won't work. The GPIO on my RPi Zero W is working, nothing with hardware problems. When i tried to ssh to it at least without the screen, it won't ping. I can't ssh to it, i've set up everything in the RNDIS/Ethernet Device settings and would show up as 10.0.0.1/10.0.0.2 but it's not working. I tried the guides on YouTube and from pwnagotchi.org. (I have macOS Ventura)

**To Reproduce**
1. Steps to reproduce the behavior:
2. Install Pwnagotchi version 2.8.9 on a Raspberry Pi Zero W with a Waveshare 2.13-inch e-Paper (Tri-Color) screen.
3. Configure SSH and test it.
4. Reinstall Pwnagotchi, performing the same steps as before to reconfigure SSH without reseting the screen.
5. Attempt to connect the screen and observe if it refreshes or displays as expected.
6. Attempt to SSH into the Pi (setting up RNDIS/Ethernet Device mode as 10.0.0.1/10.0.0.2 on macOS Ventura).
7. Try to ping or access the Pi through SSH.

**Expected behavior**
I expected for it to work with the screen while the system and Pi is working.

**Environment (please complete the following information):**
 - Pwnagotchi version 2.8.9
 - OS version: macOS Ventura 13.7
 - Type of hardware: Pi Zero W, Waveshare HAT E-Ink V4 Tri-Color display
 - Any additional hardware used: none

**Additional context**
I know the tri-color version is laggy but it worked first time, so I'm curious. 

thanks for all help!

markpavlenko, 2024-11-09:
I'm a noob in pwnagotchis please be kind

jayofelony, 2024-11-09:
Do a reflash, and put a `config.toml` file in the boot folder which only has the following, edit the display type. Nothing else, dont manually add any plugins or whatever. Also use my wiki to set it up. I don't have a config wiki for mac, but the rest should be straightforward.

```
ui.display.enabled = true
ui.display.type = "display type"
```

Just to make sure it works

markpavlenko, 2024-11-10:
I tried, but the screen is still frozen and not working
[image]

jayofelony, 2024-11-10:
What display type are you using?

closed 2024-11-15 by markpavlenko