jayofelony/pwnagotchi #238: Bettercap web UI

MimoAW, 2024-12-03:
**Describe the bug**
I can not connect to the bettercap web UI if I refreshed it while the pwnagotchi is rebooting I tried on virtual machines and different computers and different web browsers and it just won't let me in the pwnagotchi web UI does work though could there be a way to fix it in 2.9.2?
**To Reproduce**
Steps to reproduce the behavior:
1. Press refresh multiple times while the pwnagotchi is rebooting

**Expected behavior**
I can use the web UI again

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version = 2.8.9
 - OS version = Mac Sequoia latest, kali latest, fedora latest and windows 10 latest
 - Type of hardware =Pi 4b
 - Any additional hardware used =None

Thank you!

MimoAW, 2024-12-03:
Okay thank you so much!

MimoAW, 2024-12-04:
Sorry to reopen this but I just noticed yesterday, I made a typo I'm not on 2.8.2 I'm on 2.8.9 and it still doesn't work should I just wait until 2.9.2? Thank you!

jayofelony, 2024-12-04:
When you can see http://10.0.0.2:8080 can you see http://10.0.0.2:80 ?

Can you show your config.toml, because bettercap ui works, just not for you. Out of thousands of users, you are the only one experiencing this issue.

MimoAW, 2024-12-04:
No i cant see http://10.0.0.2:80 I don't know if the photos 
in the other comment will help. He is my config

main.name = "MilosPwnagotchi"
main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.mac = "D4:3A:2C:BF:03:3A"
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 500
main.plugins.bt-tether.devices.android-phone.share_internet = true
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = false
main.plugins.bt-tether.devices.ios-phone.search_order = 2
main.plugins.bt-tether.devices.ios-phone.mac = ""
main.plugins.bt-tether.devices.ios-phone.ip = "172.20.10.6"
main.plugins.bt-tether.devices.ios-phone.netmask = 24
main.plugins.bt-tether.devices.ios-phone.interval = 5
main.plugins.bt-tether.devices.ios-phone.scantime = 20
main.plugins.bt-tether.devices.ios-phone.max_tries = 0
main.plugins.bt-tether.devices.ios-phone.share_internet = true
main.plugins.bt-tether.devices.ios-phone.priority = 999

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.fix_services.enabled = true

main.plugins.gdrivesync.enabled = false
main.plugins.gdrivesync.backupfiles = [ "",]
main.plugins.gdrivesync.backup_folder = "PwnagotchiBackups"

main.plugins.gpio_buttons.enabled = false

main.plugins.gps.enabled = true
main.plugins.gps.speed = 9600
main.plugins.gps.device = "/dev/ttyACM0"

main.plugins.grid.enabled = true
main.plugins.grid.report = true

main.plugins.logtail.enabled = true
main.plugins.logtail.max-lines = 10000

main.plugins.memtemp.enabled = false
main.plugins.memtemp.scale = "celsius"
main.plugins.memtemp.orientation = "horizontal"

main.plugins.net-pos.enabled = false
main.plugins.net-pos.api_key = "test"

main.plugins.onlinehashcrack.enabled = false
main.plugins.onlinehashcrack.email = ""
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false

main.plugins.pisugar2.enabled = false
main.plugins.pisugar2.shutdown = 5
main.plugins.pisugar2.sync_rtc_on_boot = false

main.plugins.session-stats.enabled = true
main.plugins.session-stats.save_directory = "/var/tmp/pwnagotchi/sessions/"

main.plugins.ups_hat_c.enabled = false
main.plugins.ups_hat_c.label_on = true
[... 167 lines cut ...]
 "ble.device.characteristic.discovered",
 "wifi.client.new",
 "wifi.client.lost",
 "wifi.client.probe",
 "wifi.ap.new",
 "wifi.ap.lost",
 "mod.started",
]

fs.memory.enabled = true
fs.memory.mounts.log.enabled = true
fs.memory.mounts.log.mount = "/etc/pwnagotchi/log/"
fs.memory.mounts.log.size = "50M"
fs.memory.mounts.log.sync = 60
fs.memory.mounts.log.zram = true
fs.memory.mounts.log.rsync = true

fs.memory.mounts.data.enabled = true
fs.memory.mounts.data.mount = "/var/tmp/pwnagotchi"
fs.memory.mounts.data.size = "10M"
fs.memory.mounts.data.sync = 3600
fs.memory.mounts.data.zram = true
fs.memory.mounts.data.rsync = true

I'm so sorry about all of this

MimoAW, 2024-12-04:
Okay of course. Thank you for your help!

closed 2024-12-04 by MimoAW