jayofelony/pwnagotchi #243: A way to change the network IP?

theghostglitch, 2024-12-09:
**Is your feature request related to a problem? Please describe.**
The current problem is that some internet providers use the network masking. IE: Xfinity. So trying to connect the device to USB and setting the network adapter's IP to 10.0.0.1 will disconnect the original Network. You can access the device, but can no longer access the internet on you PC.

**Describe the solution you'd like**
A way to somehow change the network masking?
Or, if it's already possible, some instructions on how.

**Describe alternatives you've considered**
The only other solution would be to get a second router and connect it in between the computer and the main router (IE: xfinity's router)

**Additional context**
Not available.

jayofelony, 2024-12-09:
Not unless you use linux to directly change the contents of the sd card.

closed 2024-12-22 by jayofelony

Talaxin, 2024-12-28:
> _**Edit**_ Sorry, I did find a work around on the website. Although, it would be nice to have a way to change it before connecting it via usb. IE: a file on the SD card, much like how you can setup the the wifi with a wpa-supplicant.conf. Simply because you have to disable your current network adapter just so you can change the net mask on the device.

what was the solution?