jayofelony/pwnagotchi #244: [BUG]

nasiralamreeki, 2024-12-09:
**Describe the bug**
Flashing 2.9.2 results in boot loop and flashing display 
**To Reproduce**
Steps to reproduce the behavior:
1. Backup
2. Flash
3. Restore

**Expected behavior**
Pwnagotchi to boot the latest version and start scanning 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - 2.9.2
 - 32
 - PW0
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

closed 2024-12-09 by nasiralamreeki

BoloErpcom, 2024-12-10:
what did You do To repair ? mine is doin exactly the same . After update from 2.8.9 to 2.9.2.

nasiralamreeki, 2024-12-10:
> what did You do To repair ? mine is doin exactly the same . After update from 2.8.9 to 2.9.2.

I was able to get it to start scanning but somehow it's still loading AI even from the new release I'm at a loss of what's going on and hoping to see @jayofelony follow up on discord. I got the scanning to start by enabling the WiFi usb I have apparently backup didn't collect config.txt but somehow startup still showing old version and when it starts scanning the corner bottom right text says AI and I've checked the flash worked fine ...

jayofelony, 2024-12-10:
Dont use auto update, reflash

BoloErpcom, 2024-12-10:
Yup :) update broke everything. Managed to repair autostart but it broken display function. Backuped stuff and  Reflashing now.