jayofelony/pwnagotchi #245: Does it not create a config.toml any more

AlienMajik, 2024-12-10:
I am running the new 2.9.2 image on a pi 5. First time flashing it I waited about 6 mins after it started capturing networks shut it down and when I went to restore my backup there is no config.toml file in /etc/pwnagotchi so i just added mine and added custom faces and all my custom plugins. After restarting it crashes and freeze where it first left off. I have now reflashed it should i just edit the default.toml file instead of adding config.toml?

AlienMajik, 2024-12-10:
I think i might of found what the problem is once i start installing plugins pwnlog and pwnkill returns with error. bash: pwnlog: command not found and is now crashing. Also when I try sudo bluetoothctl it just says Waiting to connect to bluetoothd........

jayofelony, 2024-12-10:
No it does not, nor ever has, created a config.toml by itself. It would use defaults.toml if none is detected.

AlienMajik, 2024-12-10:
> No it does not, nor ever has, created a config.toml by itself.

I see i fixed my problem it was a custom plugin i made that was causing it to crash took it off and i now works. Thank you

closed 2024-12-10 by jayofelony