jayofelony/pwnagotchi #247: Bettercap web ui

AlienMajik, 2024-12-10:
When i put it manual mode then go to localhost. It no longer goes to the bettercap ui did something change that it no longer works like it used to?

jayofelony, 2024-12-10:
Localhost never worked, it would be http://10.0.0.2:80

Does that work?

AlienMajik, 2024-12-10:
What if I am using Bluetooth tethering on a iphone? 172.20.10.6:80 doesn't work

AlienMajik, 2024-12-10:
172.20.10-0/24 > 172.20.10.6
> help ui
ui (not running): Web User Interface.

172.20.10-0/24 > 172.20.10.6
[sys.log][inf] ui starting api.rest as a requirement for ui

172.20.10-0/24 > 172.20.10.6 >> panic: listen tcp 127.0.0.1:8081: bind: address already in use

goroutine 355 [running]:
:github.com/bettercap/bettercap/v2/modules/api_rest.(*RestAPI).Start.func1()
/usr/local/src/bettercap/modules/api_rest/api_rest.go:289 +0x1b0
created by github.com/bettercap/bettercap/v2/session. (*Sessionmodule).SetRunning in goroutine 1
/usr/local/src/bettercap/session/module.go:268 +0x1aB

AlienMajik, 2024-12-10:
[image]

jayofelony, 2024-12-10:
I'll take a look when I find the time.

jayofelony, 2024-12-11:
On my personal 02w bettercap ui works perfectly fine, when going to http://10.0.0.2

AlienMajik, 2024-12-12:
Its in manual mode and I am using a bnep network on a iphone. Still wont connect to bettercap ui

jayofelony, 2024-12-12:
LikeI said, its working fine with me using a fresh flash of 2.9.2

AlienMajik, 2024-12-13:
Yea i reflashed the new image didnt update. Tried a sifferent phone and it just wont work. I turned vpn and private relay. I am using bluetooth tethering with my iphone by the way. I noticed it in the last image as well one day i was able to access bettercap ui in manual mode and the next day i wasnt. Is there a way i can debug it myself?

jayofelony, 2024-12-13:
Have you tried through your pc?

AlienMajik, 2024-12-14:
No i prefer to use my phone but i will try on my computer today

jayofelony, 2024-12-14:
Okay, so apparently ui is fixed to 10.0.0.2, don't know why (yet).

AlienMajik, 2024-12-15:
Sorry for taking so long but yes it only works on a computer not sure why either

jayofelony, 2024-12-15:
I'll keep this open until I find a solution.

jayofelony, 2024-12-15:
```
sudo nmcli connection add con-name loopback type loopback
```

edit the following file:
`sudo nano /etc/udev/rules.d/85-nm-unmanaged.rules `

above line `LABEL="nm_unmanaged_end"` add `ENV{DEVTYPE}=="loopback", ENV{NM_UNMANAGED}="0"`

reboot device, see if that works

AlienMajik, 2024-12-17:
It did not work still same error api.rest server is starting at http://127.0.0.1:8081 is still cant access it though[image]

jayofelony, 2024-12-17:
Consider it work in progress.

VEYSkub, 2024-12-17:
> Локальный хост никогда не работал, это было бы http://10.0.0.2:80
> 
> Работает ли это?

bettercap

VEYSkub, 2024-12-17:
with any config settings, buttercup (on all devices, and pwnagotchi.local does not work). Moreover, I did 1 in 1 with a video where there was the same version, but alas. I had such a problem, but somehow it went away on its own (after 20 firmware overwrites). I tried using the console - python error. I fixed the error - everything is OK, it writes, but it doesn't work.

AlienMajik, 2024-12-18:
When putting pwnagotchi in manual mode then running ps aux | grep bettercap it stays running pwnagotchi-auto caplet
[image]

scansonic, 2024-12-21:
> My better cap web ui working fine about 2 days, and today is die ![IMG_7895](https://private-user-images.githubusercontent.com/105582727/397095414-b3232fa0-adef-414c-9dc2-29903242c2ab.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3NzM2MTksIm5iZiI6MTczNDc3MzMxOSwicGF0aCI6Ii8xMDU1ODI3MjcvMzk3MDk1NDE0LWIzMjMyZmEwLWFkZWYtNDE0Yy05ZGMyLTI5OTAzMjQyYzJhYi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjFUMDkyODM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmM4NzIzNDZlOWU0MzZlOTI1OTZlYzAyMDY5NjNjN2U5OGFmZTVlZjI3MmYyMTVhZGU1MWJjOTQ4NjMwODZiZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bdcHpcWImJViyMSTM4c8TsS6xiIDMPmoo3MI6yXh5Rc)

**_I found my mistake and restored web ui_**
**important to know:** when installing a real time clock on a raspberry pi, you need to install a real time clock module from github, for this you need to install the github package, which has apache2 in its dependencies. As a result, web ui does not work as it should
Good luck to everyone!

scansonic, 2024-12-21:
> When i put it manual mode then go to localhost. It no longer goes to the bettercap ui did something change that it no longer works like it used to?

can i see you history log and config.toml?

scansonic, 2024-12-21:
> What if I am using Bluetooth tethering on a iphone? 172.20.10.6:80 doesn't work

At Bluetooth tethering (auto and manual) only 2 ports is opened: 22 and 8080

NuscoMix, 2024-12-26:
Please tell me the default password for "buttercup "  , from version 2.9.3-2

closed 2024-12-26 by AlienMajik