jayofelony/pwnagotchi #248: [BUG] E-Ink Waveshare 2.13 v4 doesnnt work anymore

BoloErpcom, 2024-12-10:
**Describe the bug**
Screen is blank. Doesn't show antyhing. Same settings in the 2.8.9 were working without any problems with this screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash fresh new image 2.9.2
2. set in config.toml ui.display.type = "waveshare_4"
3. ...

**Expected behavior**
the same exact settings in 2.8.9 worked with version 4 of waveshare 2.13 inch display. now display does not start. service is running fine

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version 2.9.2
 - OS version - image from github
 - Type of hardware : Pi Zero 2 W with Waveshare 2.13 E-ink display v4
 - Any additional hardware used : none

**Additional context**
Add any other context about the problem here.

BoloErpcom, 2024-12-10:
this may come in handy : 

[INFO] [MainThread] : initializing waveshare v2in13_V3 display
Traceback (most recent call last):
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({17, 'J8:11', 'BOARD11', '17', 'WPI0', 'BCM17', 'GPIO17'}), pull='', row=6, col=1, interfaces=frozenset({'', 'dpi', 'spi', 'uart', 'gpio'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/pwnagotchi", line 8, in <module>
    sys.exit(pwnagotchi_cli())
             ^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/cli.py", line 315, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/display.py", line 310, in init_display
    self._implementation.initialize()
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/waveshare2in13_V3.py", line 37, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V3.epd2in13_V3 import EPD
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/v2in13_V3/epd2in13_V3.py", line 32, in <module>
    from pwnagotchi.ui.hw.libs.waveshare.epaper import epdconfig
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/epdconfig.py", line 164, in <module>
    implementation = RaspberryPi()
                     ^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/epdconfig.py", line 56, in __init__
    self.GPIO_RST_PIN = gpiozero.LED(self.RST_PIN)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/opt/.pwn/lib/python3.11/site-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'

BoloErpcom, 2024-12-10:
Debug shows that : 

[INFO] [MainThread] : initializing waveshare v2in13_V4 display
Traceback (most recent call last):
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({'J8:11', 'GPIO17', '17', 17, 'WPI0', 'BOARD11', 'BCM17'}), pull='', row=6, col=1, interfaces=frozenset({'', 'dpi', 'gpio', 'spi', 'uart'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/pwnagotchi", line 8, in <module>
    sys.exit(pwnagotchi_cli())
             ^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/cli.py", line 315, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/display.py", line 310, in init_display
    self._implementation.initialize()
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/waveshare2in13_V4.py", line 36, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4 import EPD
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/v2in13_V4/epd2in13_V4.py", line 32, in <module>
    from pwnagotchi.ui.hw.libs.waveshare.epaper import epdconfig
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/epdconfig.py", line 164, in <module>
    implementation = RaspberryPi()
                     ^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/epdconfig.py", line 56, in __init__
    self.GPIO_RST_PIN = gpiozero.LED(self.RST_PIN)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/opt/.pwn/lib/python3.11/site-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'

jayofelony, 2024-12-10:
Ah, run this `sudo systemctl stop pwnagotchi && sudo pwnagotchi --debug`

hugalafutro, 2024-12-12:
I ran into this issue when restoring backup - display would not initialize, fresh flash worked on 2 different pwny's after just setting the display

check your plugins, do a clean flash then add plugins one by one.
enable_assoc threw out error and not needed, enable_deauth not needed as the switches are now in the new autotune plugin
tweak_view needs `sudo apt install fonts-dejavu` as it throws error about missing font,
etc etc

Not restoring the whole backup but doing it file by file allowed me to see when the display stopped working which was when I copied in plugins.

Everize1, 2024-12-13:
I have exact same error log (1:1) when restarting gotchi to manual mode, but initially I don't have the screen issue as OP have

jayofelony, 2024-12-22:
New version released since, no updates on this issue as all displays are working. Closing.

closed 2024-12-22 by jayofelony