jayofelony/pwnagotchi #249: [BUG] Missing wlan0 which causes bettercap to not start

sakkarose, 2024-12-11:
**Describe the bug**
Bettercap can't start up because it couldn't find wlan0

**To Reproduce**
Steps to reproduce the behavior:
1. Power on

**Expected behavior**
Bettercap will start normally.

**Screenshots**
A repeating lines in dmesg.
[image]

Power on dmesg
[boot_dmesg.txt](https://github.com/user-attachments/files/18093742/boot_dmesg.txt)

On bettercap,
`no interface matching 'wlan0mon' found. bettercap`

On pwnagotchi,
[INFO] waiting for bettercap API to be available ...

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.9.2
 - OS version
 - Type of hardware: Raspberry Pi Zero 2
 - Any additional hardware used

**Additional context**
It can be fixed sometime by rebooting, but not everytime, sometime it takes me 3-4 times.
This rarely happens during usage, it happens mostly after boot.

jayofelony, 2024-12-11:
Bettercap is standalone from pwnagotchi, if used with my image all works just fine.

If you've changed anything please report at bettercap/bettercap github repo.

sakkarose, 2024-12-11:
My second reflash and it still happens. Prob faulty migrated config or custom plugins, I guess I will do a third one then. Tysm.

closed 2024-12-11 by sakkarose