jayofelony/pwnagotchi #251: [Bug]: motd complains about non-existing version file

hugalafutro, 2024-12-12:
### What happened?

small cosmetic issue, but everytime you log in motd complains about non-existing file

### Version

2.9.2 (64bit)

### Relevant log output

```shell
right after you put in password this is the first line
`/etc/update-motd.d/01-motd: 3: cannot open /usr/local/libs/python3.11/dist-packages/punagotchi/_version`
then motd displays normally after, with the face and pwn commands etc
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2024-12-12:
Yep, I noticed and am working on it. Use --version flag, that works.

jayofelony, 2024-12-12:
https://github.com/jayofelony/pwn-gen/commit/04a3d8a3d83d5d847c77292c53dc2aa5489d42d7

Has been fixed in [pwn-gen](https://github.com/jayofelony/pwn-gen).

closed 2024-12-12 by jayofelony