jayofelony/pwnagotchi #252: [Bug]: Stuck on Starting monitor interface ... with default install and even with changing the bettercaps passwords and config.toml passwd.

darkt1981, 2024-12-12:
### Contact Details

hendrikkooistra1981@gmail.com

### What happened?

After reinstalling my pwnagotchi zero 2 w it keep getting stuck on 

19:14:45 [INFO] [MainThread] : entering auto mode ...
19:14:45 [INFO] [WebServer] : web ui available at http://0.0.0.0:8080/
19:14:45 [INFO] [MainThread] : connecting to http://localhost:8081/api ...
19:14:45 [INFO] [MainThread] : starting monitor interface ...
19:14:51 [INFO] [MainThread] : starting monitor interface ...
19:14:57 [INFO] [MainThread] : starting monitor interface ...
19:15:02 [INFO] [MainThread] : starting monitor interface ...
19:15:08 [INFO] [MainThread] : starting monitor interface ...

i have tried the following:
- Changing nothing in the default passwords.
- Changing the default passwords and toml file.
- Reinstalled again without modified config.
- Tried in manuel and auto modes and even over bluetooth connection on phone browser.

Weird part is i can access the bettercap portal in the browser, yet the pi it self keep hanging on this.

### Version

2.9.2 (64bit)

### Relevant log output

```shell
19:14:29 [WARNING] [Dummy-48] : restarting in AUTO mode ...
19:14:37 [INFO] [MainThread] : [hashie] plugin loaded
19:14:37 [INFO] [MainThread] : [hashie] plugin loaded
19:14:37 [INFO] [Thread-20 (run_once)] : OHC: OnlineHashCrack plugin loaded.
19:14:37 [INFO] [Thread-21 (run_once)] : gps plugin loaded for /dev/ttyACM0
19:14:37 [INFO] [Thread-22 (run_once)] : [webgpsmap]: plugin loaded
19:14:37 [INFO] [Thread-24 (run_once)] : Logtail plugin loaded.
19:14:37 [INFO] [Thread-23 (run_once)] : Session-stats plugin loaded.
19:14:37 [INFO] [Thread-25 (run_once)] : grid plugin loaded.
19:14:37 [INFO] [Thread-26 (run_once)] : [update] plugin loaded.
19:14:37 [INFO] [Thread-27 (run_once)] : webcfg: Plugin loaded.
19:14:37 [INFO] [Thread-29 (run_once)] : [Fix_Services] plugin loaded.
19:14:37 [INFO] [Thread-31 (run_once)] : [HandshakesDL] plugin loaded
19:14:37 [INFO] [Thread-30 (run_once)] : [Internet Conection] Plugin loaded.
19:14:37 [INFO] [Thread-32 (run_once)] : [*] DiscoHash plugin loaded
19:14:37 [INFO] [Thread-33 (run_once)] : [HandshakesDL] plugin loaded
19:14:37 [INFO] [Thread-16] : [hashie] Starting batch conversion of pcap files
19:14:37 [INFO] [Thread-17] : [hashie] Starting batch conversion of pcap files
19:14:37 [INFO] [Thread-19 (run_once)] : BT-TETHER: Successfully loaded ...
19:14:37 [WARNING] [UI Handler] : non fatal error while updating view: 'Display' object has no attribute '_ignore_changes'
19:14:37 [INFO] [MainThread] : initializing waveshare v2in13_V3 display
19:14:43 [INFO] [MainThread] : initializing waveshare v3 display done
19:14:44 [INFO] [MainThread] : Starting WebServer thread
19:14:45 [INFO] [MainThread] : Terrorinc@ca3494c9122e8c21fe6a9ea89a37b9c1857aef2a98a8e733cd7963e9a4ea8607 (v2.9.2)
19:14:45 [INFO] [MainThread] : entering auto mode ...
19:14:45 [INFO] [WebServer] : web ui available at http://0.0.0.0:8080/
19:14:45 [INFO] [MainThread] : connecting to http://localhost:8081/api ...
19:14:45 [INFO] [MainThread] : starting monitor interface ...
19:14:51 [INFO] [MainThread] : starting monitor interface ...
19:14:57 [INFO] [MainThread] : starting monitor interface ...
19:15:02 [INFO] [MainThread] : starting monitor interface ...
19:15:08 [INFO] [MainThread] : starting monitor interface ...
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2024-12-12:
Hallo Hendrik,

Heb je wellicht een systeem upgrade gedraaid? (Apt-get update/upgrade)

Mag ik je config.toml zien, eventuele vertrouwelijke gegevens wel even afschermen.

Ik hoor het wel.

darkt1981, 2024-12-12:
nee had een oude config van 1.5.5
main.iface = "mon0" moest veranderd worden naar main.iface = "wlan0mon"

closed 2024-12-12 by darkt1981

jayofelony, 2024-12-12:
Vermoeden had ik al.