jayofelony/pwnagotchi #254: [Bug]: No wifi in 2.9.2 32bit

htotoo, 2024-12-13:
### What happened?

Just flashed a fresh install to my raspi zero w, and there is no mon0 or wlan0, bettercap says it can't start monitor mode, ...
I flashed evilsocket/pwnagotchi, and it started to work.
Tested without any external wifi, just tried to use the internal one.

### Version

2.9.2 (32bit)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

htotoo, 2024-12-13:
Flashed the v2.8.9, and it seems to work.

htotoo, 2024-12-13:
That is the default, and that was set. I ssh-d into it, and cant see any devices. with 2.8.9 also ssh-d to it, and with iwconfig i see wlan0, wlan0mon too. from the 2.9.2, those were missing (in every mode)

jayofelony, 2024-12-13:
Did you flash or auto update?

htotoo, 2024-12-13:
Flashed a fresh one to a new sd.

htotoo, 2024-12-14:
Just upgraded from 2.8.9 to 2.9.2 via self updater, and still working. Seems like only fresh install don't work.

jayofelony, 2024-12-14:
Fresh flash works as it should, just not for you. Auto update should not be used.

closed 2024-12-14 by htotoo