jayofelony/pwnagotchi #255: Send terminal commands from WebUI

michaelbapst, 2024-12-14:
I was wondering if there was a way to send terminal commands through the WebUI. This would be a real help when Pwnagotchi is connected to the phone via bluetooth. Or a button at the bottom of the WebUI that basically performs the same function as 
`pwnagotchi --clear`
either would be great.

wpa-2, 2024-12-14:
whats wrong with using terminal ? 
apple/android both have options.

michaelbapst, 2024-12-15:
I haven't been able to ssh with a terminal app via bluetooth, I figured it wasn't possible because bluetooth can be weird like that. Still would make a nice enhancement, be able to do simple terminal commends from the WebUI.

wpa-2, 2024-12-15:
> I haven't been able to ssh with a terminal app via bluetooth, I figured it wasn't possible because bluetooth can be weird like that. Still would make a nice enhancement, be able to do simple terminal commends from the WebUI.

ssh over bluetooth works without issues, you just use the bluetooth IP to connect.

jayofelony, 2024-12-22:
Use a simple as like Termius for that. Closing.

closed 2024-12-22 by jayofelony