jayofelony/pwnagotchi #258: [Bug]: WaveShare 3.5LCD not working

Pittsie72, 2024-12-17:
### Contact Details

spittsie72@protonmail.com

### What happened?

Been about a year since I plugged mine in, noticed the major update so thought I'd flash the latest and play around. Went through the --wizard set it up, rebooted and still just a blank white screen :(

### Version

2.9.2 (64bit)

### Relevant log output

```shell
sudo pwnagotchi --debug
[2024-12-17 14:37:32,653] [DEBUG] [File Sys] : [FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
[2024-12-17 14:37:32,694] [DEBUG] [MainThread] : loading plugins from /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default - enabled: ['bt-tether', 'auto-tune', 'auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-12-17 14:37:32,695] [DEBUG] [MainThread] : loading /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/bt-tether.py
[2024-12-17 14:37:32,714] [DEBUG] [MainThread] : loaded plugin bt-tether as <bt-tether.BTTether object at 0x7fa6f95c10>
[2024-12-17 14:37:32,715] [DEBUG] [MainThread] : PLUGIN EVENT QUEUE FOR bt-tether starting None
[2024-12-17 14:37:32,715] [DEBUG] [Thread-1] : Worker thread starting for bt-tether
[2024-12-17 14:37:32,716] [DEBUG] [MainThread] : loading /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/session-stats.py
[2024-12-17 14:37:32,718] [DEBUG] [MainThread] : loaded plugin session-stats as <session-stats.SessionStats object at 0x7fa6f96ad0>
[2024-12-17 14:37:32,719] [DEBUG] [MainThread] : PLUGIN EVENT QUEUE FOR session-stats starting None
[2024-12-17 14:37:32,719] [DEBUG] [Thread-3] : Worker thread starting for session-stats
[2024-12-17 14:37:32,720] [DEBUG] [MainThread] : loading /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/grid.py
[2024-12-17 14:37:32,721] [DEBUG] [MainThread] : loaded plugin grid as <grid.Grid object at 0x7fa6fa00d0>
[2024-12-17 14:37:32,722] [DEBUG] [MainThread] : PLUGIN EVENT QUEUE FOR grid starting None
[2024-12-17 14:37:32,723] [DEBUG] [Thread-4] : Worker thread starting for grid
[2024-12-17 14:37:32,724] [DEBUG] [MainThread] : loading /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/auto-update.py
[2024-12-17 14:37:32,725] [DEBUG] [MainThread] : loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7fa6fa1890>
[2024-12-17 14:37:32,726] [DEBUG] [MainThread] : PLUGIN EVENT QUEUE FOR auto-update starting None
[2024-12-17 14:37:32,727] [DEBUG] [Thread-5] : Worker thread starting for auto-update
[2024-12-17 14:37:32,727] [DEBUG] [MainThread] : loading /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/webcfg.py
[2024-12-17 14:37:32,728] [DEBUG] [MainThread] : loaded plugin webcfg as <webcfg.WebConfig object at 0x7fa6fa25d0>
[2024-12-17 14:37:32,729] [DEBUG] [MainThread] : PLUGIN EVENT QUEUE FOR webcfg starting None
[2024-12-17 14:37:32,729] [DEBUG] [Thread-6] : Worker thread starting for webcfg
[2024-12-17 14:37:32,730] [DEBUG] [MainThread] : loading /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/auto-tune.py
[2024-12-17 14:37:32,732] [DEBUG] [MainThread] : loaded plugin auto-tune as <auto-tune.auto_tune object at 0x7fa6fb1110>
[2024-12-17 14:37:32,732] [DEBUG] [MainThread] : PLUGIN EVENT QUEUE FOR auto-tune starting None
[2024-12-17 14:37:32,733] [DEBUG] [Thread-7] : Worker thread starting for auto-tune
[2024-12-17 14:37:32,733] [DEBUG] [MainThread] : loading /opt/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/fix_services.py
[2024-12-17 14:37:32,736] [DEBUG] [MainThread] : loaded plugin fix_services as <fix_services.FixServices object at 0x7fa6fb2a90>
[2024-12-17 14:37:32,737] [DEBUG] [MainThread] : PLUGIN EVENT QUEUE FOR fix_services starting None
[2024-12-17 14:37:32,738] [DEBUG] [Thread-8] : Worker thread starting for fix_services
[2024-12-17 14:37:32,739] [DEBUG] [MainThread] : loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['bt-tether', 'auto-tune', 'auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-12-17 14:37:32,740] [DEBUG] [Thread-9 (run_once)] : bt-tether.loaded: ()
[2024-12-17 14:37:32,740] [DEBUG] [MainThread] : bt-tether on_loaded
[2024-12-17 14:37:32,741] [DEBUG] [Thread-10 (run_once)] : session-stats.loaded: ()
[2024-12-17 14:37:32,741] [DEBUG] [MainThread] : session-stats on_loaded
[2024-12-17 14:37:32,743] [DEBUG] [Thread-11 (run_once)] : grid.loaded: ()
[2024-12-17 14:37:32,743] [INFO] [Thread-11 (run_once)] : grid plugin loaded.
[2024-12-17 14:37:32,743] [DEBUG] [MainThread] : grid on_loaded
[2024-12-17 14:37:32,744] [INFO] [Thread-10 (run_once)] : Session-stats plugin loaded.
[2024-12-17 14:37:32,744] [DEBUG] [Thread-11 (run_once)] : Thread for grid loaded exiting
[2024-12-17 14:37:32,745] [DEBUG] [Thread-12 (run_once)] : auto-update.loaded: ()
[2024-12-17 14:37:32,745] [DEBUG] [MainThread] : auto-update on_loaded
[2024-12-17 14:37:32,745] [DEBUG] [Thread-10 (run_once)] : Thread for session-stats loaded exiting
[2024-12-17 14:37:32,745] [INFO] [Thread-12 (run_once)] : [update] plugin loaded.
[2024-12-17 14:37:32,746] [DEBUG] [Thread-13 (run_once)] : webcfg.loaded: ()
[2024-12-17 14:37:32,747] [INFO] [Thread-13 (run_once)] : webcfg: Plugin loaded.
[2024-12-17 14:37:32,747] [DEBUG] [MainThread] : webcfg on_loaded
[2024-12-17 14:37:32,747] [DEBUG] [Thread-12 (run_once)] : Thread for auto-update loaded exiting
[2024-12-17 14:37:32,747] [DEBUG] [Thread-13 (run_once)] : Thread for webcfg loaded exiting
[2024-12-17 14:37:32,748] [DEBUG] [Thread-14 (run_once)] : auto-tune.loaded: ()
[2024-12-17 14:37:32,749] [DEBUG] [MainThread] : auto-tune on_loaded
[2024-12-17 14:37:32,749] [DEBUG] [Thread-14 (run_once)] : Thread for auto-tune loaded exiting
[2024-12-17 14:37:32,750] [DEBUG] [Thread-15 (run_once)] : fix_services.loaded: ()
[2024-12-17 14:37:32,750] [DEBUG] [MainThread] : fix_services on_loaded
[... 8 lines cut ...]
[2024-12-17 14:37:32,769] [WARNING] [MainThread] : ui.fps is 0, the display will only update for major changes
[2024-12-17 14:37:32,769] [DEBUG] [Thread-8] : Got here
[2024-12-17 14:37:32,772] [INFO] [MainThread] : initializing waveshare 3,5inch lcd display
Traceback (most recent call last):
  File "/usr/bin/pwnagotchi", line 8, in <module>
    sys.exit(pwnagotchi_cli())
             ^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/cli.py", line 315, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/display.py", line 310, in init_display
    self._implementation.initialize()
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/waveshare3in5lcd.py", line 42, in initialize
    self._display.ready_fb(i=0)
  File "/opt/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/libs/fb/fb.py", line 46, in ready_fb
    with open('/dev/fb' + str(i), 'r+b') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/fb0'
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2024-12-17:
`sudo apt-get install fbi`?

I think there is some installation somewhere how to set the 3.5 inch LCD up.

Pittsie72, 2024-12-17:
Hmmm okay let me keep digging. Thought it would work out of the box since the "displays" link in the --wizard included it

jayofelony, 2024-12-17:
The display files are there, but it still needs drivers to be installed.

jayofelony, 2024-12-22:
Assuming this is resolved. Closing.

closed 2024-12-22 by jayofelony