jayofelony/pwnagotchi #260: New and no Idea

Penetrationpasi-hub, 2024-12-17:
Hi, I just bought a Pwnagotchi and downloaded the latest image. However, when I plug the Pwnagotchi into my Windows 11 (64-bit) PC, nothing happens, even if I wait more than 10 minutes. There should be an Ethernet interface appearing, right? Why isn't that happening? I’m using an RPi Zero 2 W with a Waveshare 2.13 V4 display. I used Raspberry Pi OS as described in the setup guide. I also downloaded the latest RNDIS driver from https://pwnagotchi.org/getting-started/software/index.html, but when I try to install it, Windows tells me the best driver is already installed, but nothing happens.

jayofelony, 2024-12-17:
Most probable cause is the USB cable used not being able to send data.

Pastel1130, 2024-12-24:
I'm having a similar problem but I have a data cable

wpa-2, 2024-12-24:
> I'm having a similar problem but I have a data cable

Uninstall the current driver and then install the mod version.

NuscoMix, 2024-12-26:
I had the same problem, I put a microUSB - USB Type C cable. The connection problem is gone

Pastel1130, 2024-12-27:
Yeah I'm having the same issue. The Pwan comes on and works, but doesn't come up in Device Manager as anything.

jayofelony, 2024-12-28:
Closing this as this is not a software related issue to Pwnagotchi, most cables advertised as data cable are actually not a data cable.

closed 2024-12-28 by jayofelony

Penetrationpasi-hub, 2024-12-28:
Hi,

no, it's not the data cable. I used an adapter to USB-C, and I was able to transfer as much data as I wanted.
It also seems that I'm not the only one struggling with this. So, where should I ask this question instead?

My Windows PC just doesn't recognize the Pwnagotchi, no matter what I do. I tried the setup guide three times.
There isn't even the usual sound when you plug in an USB Device, just  nothing happens.

LG Pascal Range

Pastel1130, 2024-12-28:
@jayofelony could you provide us with a link to a data cable that works?