jayofelony/pwnagotchi #262: Anyone else having issue using external antennae?

allordacia, 2024-12-20:
After editing the new RPI config file to disable onboard wifi (pi0w2) I cannot get a previously working USB dongle (panda wireless) to be usable. The pwnagotchi config file no longer has an option to specify the interface which is likely the issue but was not sure if I could manually add it back or if that would cause any other issues?

renewedbullet, 2024-12-21:
What image are you using?

jayofelony, 2024-12-28:
Closing, stale issue.

closed 2024-12-28 by jayofelony

rezzorix, 2025-02-08:
> Closing, stale issue.

Not OP, but I want to reopen this. I think the question could have been answered as it was clear.

Anyway, here we go:
Version: 2.9.5.3

I have the same issue, I do not want to use the Raspberry 3/4 onboard wifi - I want to use an USB dongle.

What files / settings do I have to modify?

Just disabling the dtoverlay wifi in the config.txt will result in reboot loops...

Any advise?

wpa-2, 2025-02-08:
> Not OP, but I want to reopen this. I think the question could have been answered as it was clear.
>
> Anyway, here we go: Version: 2.9.5.3
>
> I have the same issue, I do not want to use the Raspberry 3/4 [...]

Not for me all I did was comment out
dtoverlay line related to your raspberry pi

/boot/firmware/config.txt

The only reason it won't work is if your dongle isn't compatible..

rezzorix, 2025-02-08:
ok... so fresh install.

Hardware: 
- Raspberry Pi3
- MediaTek Inc. MT7612U

Steps
0. fresh install
1. I disable wifi by uncommenting dtoverlay=disable-wifi
2. I reboot
3. the wifi stick is compatible and working; it identifies APs.
4. problem seems to be somehow with bettercap:

```
17:00:02 [INFO] [MainThread] : sending association frame to <hidden> (a6:2a:95:bd:fd:d5 ) on channel 6 [0 clients], -70 dBm...
17:00:04 [INFO] [MainThread] : sending association frame to EMBUNKLCC_2.4G (0a:4f:66:10:7c:c1 ) on channel 6 [0 clients], -69 dBm...
17:00:05 [INFO] [MainThread] : sending association frame to Angel_2.4 (a4:2a:95:bd:fd:d5 D-Link International) on channel 6 [0 clients], -69 dBm...
17:00:05 [WARNING] [Event Polling] : [bettercap] ping error - retrying connection in 4.563169228909471 sec
17:00:05 [INFO] [MainThread] : sending association frame to 328-12-03 (0c:80:63:5a:d6:1c TP-LINK TECHNOLOGIES CO.,LTD.) on channel 6 [0 clients], -70 dBm...
17:00:05 [WARNING] [MainThread] : [bettercap] can't run my request... connection to the bettercap endpoint failed...
17:00:05 [WARNING] [MainThread] : [bettercap] retrying run in 4.451515176596059 sec
17:00:07 [ERROR] [Session Fetcher] : [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='127.0.0.1', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fac0f7d10>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
```

After running through more like above the system will reboot.. and loop..

allordacia, 2025-02-08:
Yeah I never got this working and the version originally mentioned not the newest. I tried with two different dongles that are compatible with my other pwnagotchi devices and neither were successful. In all cases, it is unable to start bettercap according to the logs and enters into the boot loop mentioned previously.

rezzorix, 2025-02-08:
Ok, tried now with editing the config.txt and disabling wifi "before" the first system boot.

Same problem.

> Yeah I never got this working on the version originally mentioned nor the latest. I tried with two different dongles that are compatible with my other pwnagotchi devices and neither were successful. In all cases, it is unable to start bettercap according to the logs and enters into the boot loop mentioned previously.

Yes, problem is with bettercap somehow.

@jayofelony any thoughts / suggestions?

wpa-2, 2025-02-08:
If your using external make sure you ahve fix_services off as its only for internal wifi.

allordacia, 2025-02-08:
> If your using external make sure you ahve fix_services off as its only for internal wifi. 

Good reminder but I can confirm that I had it disabled while testing both of a panda wireless n600 and Brotrend adapters, both which work fine on a unit running the last "official" pwnagotchi firmware.

wpa-2, 2025-02-08:
> > If your using external make sure you ahve fix_services off as its only for internal wifi.
> 
> Good reminder but I can confirm that I had it disabled while testing a panda wireless n600 and a generic Brotrend adapter, both which work fine on a unit running the last "official" pwnagotchi firmware.

latest official?

rezzorix, 2025-02-09:
> > If your using external make sure you ahve fix_services off as its only for internal wifi.
> 
> Good reminder but I can confirm that I had it disabled while testing a panda wireless n600 and a generic Brotrend adapter, both which work fine on a unit running the last "official" pwnagotchi firmware.

This is it. 
Switch off "fix_services" and all works fine!