jayofelony/pwnagotchi #263: No address from dhcp

czadbastian, 2024-12-21:
### What happened?

I have a rpi 4 8Gb. No address from dhcp. No entry in the arp table either. Two device restarts and the same effect. Does anyone have something similar?

### Version

2.9.2 (64bit)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2024-12-21:
Being worked on. Next update

jayofelony, 2024-12-22:
https://github.com/jayofelony/pwn-gen/commit/1c9fa179b435dc25656ea337e2ace17055188ac3
https://github.com/jayofelony/pwn-gen/commit/dffe8900dc555276cdfb7407258c0b8682788226
https://github.com/jayofelony/pwn-gen/commit/d8ddb6a1ef0d29bab5d1c3cc5afacc99c48aa7e6
https://github.com/jayofelony/pwn-gen/commit/5b1f1a84e75b76002e7f1f00afc8bacf287b0915

Fixed for next release.

closed 2024-12-22 by jayofelony