jayofelony/pwnagotchi #265: PisugarX Plugin: How to Limit Percentage to 2 Decimal Places?

languagemaniac, 2024-12-21:
Hi,

The battery percentage displayed on my Pwnagotchi shows too many decimal places, causing it to collide with the "UP" label. Which didn't happen before updating so I'm guessing it's because of the pisugarX plugin.

For example, it currently displays: 74.88407%, and sometimes the number is even longer, making it look messy.

I’d like to limit the percentage to 2 decimal places for a cleaner display. How can I do this?

Thanks!

wpa-2, 2024-12-21:
I believe this is going to be looked at in the next release, but if you cant wait you will need to edit the plugin.

jayofelony, 2024-12-22:
https://github.com/jayofelony/pwnagotchi/commit/0472faa2bf536a0e0938bacc7094c15c3bd50692

Fixed.

closed 2024-12-22 by jayofelony