jayofelony/pwnagotchi #268: No plugins 2.9.3 image

AlienMajik, 2024-12-22:
It no longer has a available plugins or custom plugins: sudo pwnagotchi plugins list does not work. Pwnkill does not work either. Also there are 2 handshake folders being created one in /root and the other at /home/pi

wpa-2, 2024-12-22:
[image]

default folder now is /home/pi/handshakes/ 

pwnkill works just fine. 
[image]

check debug 
```
sudo systemctl stop pwnagotchi.service
sudo pwnagotchi --debug 
```

wpa-2, 2024-12-22:
@jayofelony  looks like this has been sorted maybe.

AlienMajik, 2024-12-23:
Is this on the 2.9.3 image?

wpa-2, 2024-12-23:
> Is this on the 2.9.3 image?

Yes it is.

AlienMajik, 2024-12-23:
Not sure what happened then it just would not work for so i just reflashed back to 2.9.2 and it started working

wpa-2, 2024-12-23:
> Not sure what happened then it just would not work for so i just reflashed back to 2.9.2 and it started working

try the new image https://github.com/jayofelony/pwnagotchi/releases/tag/v2.9.3-2

this can be closed though.

Agnar319, 2024-12-23:
I've flashed probably 10 times, with 4 different SDs on 2 different Pis and can't, for the life of me, get plugins to work either. Also, probably just a me issue, but i cant get output to any of my displays or web ui.

wpa-2, 2024-12-23:
> I've flashed probably 10 times, with 4 different SDs on 2 different Pis and can't, for the life of me, get plugins to work either. Also, probably just a me issue, but i cant get output to any of my displays or web ui.

As its a different issue please create your own. 
check debug, but remember a lot of the plugins will be outdated now as there's been a good few updates/changes.

AlienMajik, 2024-12-24:
Getting same errors on 2.9.3-2 image. After rebooting most files disappeared going to reflash and run it for awhile and see what happens[image][image]

AlienMajik, 2024-12-24:
Yea thats weird after flashing the second time it is now working? How who knows? Problem fixed

wpa-2, 2024-12-24:
The wiki clear states the user PI is needed for the new versions due to permission issues. 
You are going to need to reflash.

AlienMajik, 2024-12-24:
What do you mean by that? When I first boot and it asks me to give the host name just put in pi instead of a custom name?

wpa-2, 2024-12-24:
> What do you mean by that? When I first boot and it asks me to give the host name just put in pi instead of a custom name?

I'm talking about username not host name. 
Please check the wiki and see the screenshots showing the settings you need on RPI imager.

AlienMajik, 2024-12-24:
Ok I did that named it pi and it only works with my AWUS036AXM and not with the stock wifi module of my pi 5 os it because it can now scan for 6ghz wifi networks?

AlienMajik, 2024-12-24:
It is now working on both it just takes awhile for it to activate the tx power on the stock pi 5 wifi module

closed 2024-12-24 by AlienMajik