jayofelony/pwnagotchi #269: [Bug]: bluetooth web ui

saurabhkumar1346, 2024-12-22:
### Contact Details

saurabhkumar1346@gmail.com

### What happened?

web ui working fine in pc 10.0.0.2 but in bluetooth cant access web ui at 192.168.44.4 (raspberry pi zero w 32bit)
[image]

### Version

2.9.2 (32bit)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wpa-2, 2024-12-22:
https://github.com/jayofelony/pwnagotchi/wiki/Step-4-Customization#bt-tethering

Your config is wrong,  check out the wiki.

renewedbullet, 2024-12-22:
https://github.com/jayofelony/pwnagotchi/wiki/Step-4-Customization

closed 2024-12-22 by jayofelony