jayofelony/pwnagotchi #271: BLUE SCREEN OF DEATH

PRADEEPDHIBAR, 2024-12-22:
When i connect my raspberry pi zero 2w to pc for ethernet . It's recognised the driver rndis but always goes to blue screen of death.
 Window 11 pro
Raspberry pi zero 2w 

Is there any solution for this?

wpa-2, 2024-12-22:
When you say ethernet us that via and usb eth adapter or a straight usb cable.

PRADEEPDHIBAR, 2024-12-22:
Straight usb cable

wpa-2, 2024-12-22:
> Straight usb cable

Good chance it's the usb driver you have installed.

PRADEEPDHIBAR, 2024-12-22:
> > Straight usb cable
> 
> Good chance it's the usb driver you have installed.

 Rndis driver was pre-installed with Windows in my PC.

wpa-2, 2024-12-22:
Install the MOD driver that's on the wiki.

wpa-2, 2024-12-22:
> > Install the MOD driver that's on the wiki.
> 
> I have also installed that but it is not getting installed, it is written in it that the best driver is already installed. https://modclouddownloadprod.blob.core.windows.net/shared/mod-duo-rndis.zip

Maybe uninstall it first.

PRADEEPDHIBAR, 2024-12-22:
> > > Install the MOD driver that's on the wiki.
> > 
> > 
> > I have also installed that but it is not getting installed, it is written in it that the best driver is already installed. https://modclouddownloadprod.blob.core.windows.net/shared/mod-duo-rndis.zip
> 
> Maybe uninstall it first.

How do I uninstall it? I don't even get time to uninstall it. As soon as I open the device manager, the blue screen of death appears.

wpa-2, 2024-12-22:
That's not really an issue for this github, but go to device manager find it and uninstall you may need to view hidden devices.

PRADEEPDHIBAR, 2024-12-22:
> That's not really an issue for this github, but go to device manager find it and uninstall you may need to view hidden devices.

I have tried 20 times to get at least some time to open the device manager and uninstall it. But every time the blue screen of death comes immediately

PRADEEPDHIBAR, 2024-12-22:
> That's not really an issue for this github, but go to device manager find it and uninstall you may need to view hidden devices.

Ok, I uninstalled the driver and after that the PC was not going into BSOD, but as soon as I reinstalled the mod driver, the PC again started going into BSOD

wpa-2, 2024-12-22:
> > That's not really an issue for this github, but go to device manager find it and uninstall you may need to view hidden devices.
> 
> I have tried 20 times to get at least some time to open the device manager and uninstall it. But every time the blue screen of death comes immediately

dont plugin the pizero then

PRADEEPDHIBAR, 2024-12-22:
> dont plugin the pizero then

What should I do now? I uninstalled and reinstalled the driver.

wpa-2, 2024-12-22:
try plugging in the pizero and also try a different slot.

PRADEEPDHIBAR, 2024-12-22:
Thanks now working

wpa-2, 2024-12-22:
> Thanks now working

Please now its sorted.

closed 2024-12-22 by PRADEEPDHIBAR