jayofelony/pwnagotchi #272: [Bug]: No Eth0 internet access, even though SSH is working

flippinsweetdude, 2024-12-22:
### Contact Details

aaron@flippinsweetdude.com

### What happened?

Fresh install of hot fix version [2.9.3-2](https://github.com/jayofelony/pwnagotchi/releases/tag/v2.9.3-2)

No modifications / installs / custom plugins or any changes.   Plugged in RasPi 4 into network and boot up.   RasPi was able to get IP address successfully, and SSH with default user/pass works successfully.

command line : ping 8.8.8.8

Results : 
[image]

Note this routes out from 10.0.0.2

command line : sudo netstat -nr

[image]

Note line 3 has no gateway 

command line :  sudo nmcli connection down usb0

[image]

then command line : sudo netstat -nr

Results : 
[image]

Then ping 8.8.8.8

Results : 
[image]

### Version

2.9.3-2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2024-12-23:
> Note line 3 has no gateway

It has, it's the top one. It's a matter of wrong routing I think. I'll see what I can do.

flippinsweetdude, 2024-12-23:
output of IP -4 route is : 

[image]

jayofelony, 2024-12-23:
`sudo nmcli connection modify eth0 ip4.route-metric 100`

try this, see if that helps

flippinsweetdude, 2024-12-23:
Error: invalid or not allowed setting 'ip4': 'ip4' not among [connection, 802-3-ethernet (ethernet), 802-1x, dcb, sriov, ethtool, match, ipv4, ipv6, hostname, tc, proxy].

jayofelony, 2024-12-23:
Yeah, typo -> should be ipv4. edited previous comment.

flippinsweetdude, 2024-12-23:
should be ipv4

flippinsweetdude, 2024-12-23:
[image]

jayofelony, 2024-12-23:
also run:

`sudo nmcli connection reload && sudo systemctl restart NetworkManager` afterwards.

flippinsweetdude, 2024-12-23:
That worked.

flippinsweetdude, 2024-12-23:
Perfect

[image]

flippinsweetdude, 2024-12-23:
ip -4 route : 

[image]

jayofelony, 2024-12-23:
Ok, I'll add the route-metric to the connection profiles.

closed 2024-12-23 by jayofelony