jayofelony/pwnagotchi #273: Not able to pair BT as before - getting Failed to connect to device: Command '['nmcli', 'device', 'connect', '']'

Miggzlc, 2024-12-23:
As stated this is the error I see when I check logtail

"[ERROR] | [Thread-1] : [BT-Tether] Failed to connect to device: Command '['nmcli', 'device', 'connect', '']' returned non-zero exit status 10". 
I've un trusted and unpaired my phone and attempted to manually pair it a few times in which it would work on the previous firmware. I'm using a S23 Ultra if that is relevant. Please let me know what else anyone would need to know. 

my config file is:

main.name = "pwny"
main.whitelist = [
 "XXXX"
]
main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = "XXXXX"
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 14
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 20
main.plugins.bt-tether.devices.android-phone.max_tries = 0
main.plugins.bt-tether.devices.android-phone.share_internet = true
main.plugins.bt-tether.devices.android-phone.priority = 99

main.plugins.bt-tether.mac = ""
main.plugins.bt-tether.phone = ""
main.plugins.bt-tether.ip = ""

main.plugins.auto-tune.enabled = true
main.plugins.auto-tune.show_hidden = false
main.plugins.auto-tune.reset_history = true
main.plugins.auto-tune.extra_channels = 3

main.plugins.auto-update.enabled = false
main.plugins.auto-update.install = false
main.plugins.auto-update.interval = 1

main.plugins.fix_services.enabled = true

main.plugins.gdrivesync.enabled = false
main.plugins.gdrivesync.backupfiles = [ "",]
main.plugins.gdrivesync.backup_folder = "PwnagotchiBackups"

main.plugins.gpio_buttons.enabled = false

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.gps_listener.enabled = false

main.plugins.grid.enabled = true
main.plugins.grid.report = true

main.plugins.logtail.enabled = true
main.plugins.logtail.max-lines = 10000

main.plugins.memtemp.enabled = false
main.plugins.memtemp.scale = "celsius"
main.plugins.memtemp.orientation = "horizontal"

main.plugins.onlinehashcrack.enabled = false
main.plugins.onlinehashcrack.email = ""
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false

main.plugins.pisugarx.enabled = false
main.plugins.pisugarx.rotation = false
main.plugins.pisugarx.default_display = "percentage"

main.plugins.session-stats.enabled = true
main.plugins.session-stats.save_directory = "/var/tmp/pwnagotchi/sessions/"

main.plugins.ups_hat_c.enabled = false
main.plugins.ups_hat_c.label_on = true
main.plugins.ups_hat_c.shutdown = 5
main.plugins.ups_hat_c.bat_x_coord = 140
main.plugins.ups_hat_c.bat_y_coord = 0

main.plugins.ups_lite.enabled = false
main.plugins.ups_lite.shutdown = 2

main.plugins.webcfg.enabled = true

main.plugins.webgpsmap.enabled = false

main.plugins.wigle.enabled = false
main.plugins.wigle.api_key = ""

PRADEEPDHIBAR, 2024-12-23:
I was also facing the same problem. But when I edited my config in this format, the issue got fixed .

main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.phone = "" # android or ios
main.plugins.bt-tether.mac = "" # your phone's MAC address
main.plugins.bt-tether.ip = "" # 192.168.44.2-254 for android, 172.20.10.2-254 for ios

siklosi, 2024-12-23:
I got connected with those new lines but internet sharing through phone is not working (with 2.9.2 was working). Also i configured 192.168.44.44 but pwnagotchi is at 192.168.44.248.

jayofelony, 2024-12-23:
Read the wiki, closing this now.

closed 2024-12-23 by jayofelony