jayofelony/pwnagotchi #274: [Bug]: USB dongles are not working

Delurkdotcom, 2024-12-24:
### What happened?

Hi,
I am using a Raspberry Pi Zero 2W with a Waveshare 4 screen, all powered by a PiSugar3. Everything works fine without any dongle. However, when I try to use my dongles separately, either they are not detected by the `lsusb` command, or the Pwnagotchi gets stuck on "Reading last session logs" during startup, which requires a firmware reflash to restore functionality.  

The dongles I am trying to use:  
- Alfa AWUS036AXM  
- VK-172 GPS Ublox7  

The configurations used:  
- `dtoverlay=disable-wifi` in `config.txt`  
-  
```
main.plugins.gps.enabled = true
main.plugins.gps.speed = 9600
main.plugins.gps.device = "/dev/ttyUSB0"
```  
in `config.toml`.  

### Version

2.9.3-2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wpa-2, 2024-12-24:
Your gps device is wrong it won't be /dev/ttyUSB it should be something like ACM0 maybe but if you search you will find it.

Delurkdotcom, 2024-12-24:
Yes, that was an example configuration, but in any case, I don't see it listed with `lsusb`, and it's also not present in `/dev/`.

wpa-2, 2024-12-24:
> Yes, that was an example configuration, but in any case, I don't see it listed with `lsusb`, and it's also not present in `/dev/`.

How are you connecting both devices and which socket are you using?

wpa-2, 2024-12-24:
[image]
As you can see the pi can see the devices connected.

Delurkdotcom, 2024-12-24:
In the photo, you can see the cables I am using; they are USB female to micro-USB cables. In the terminal screenshot, `lsusb` does not list the Alfa interface connected to the micro-USB port, while the other port is being used for communication via my PC's USB interface over SSH.
[image]
[image]

wpa-2, 2024-12-24:
Your trying to get data from a power only port, your going to need a multi adaptor for this to work.

Usn dongles only work via the left isb slot.

Delurkdotcom, 2024-12-24:
So, it's just a hardware issue then? Thank you, it seems so simple. Do I need a hub? A self-powered hub? A multiport adapter? Could you provide an example, please?

jayofelony, 2024-12-24:
Closing this as it's clearly user error, no bug whatsoever. Open a discussion to talk about hardware.

closed 2024-12-24 by jayofelony

Delurkdotcom, 2024-12-24:
Also, how can I fix the "Reading last session logs" issue? I’ve already tried `sudo rm -R /etc/pwnagotchi/log*`, but it doesn’t work. I have to erase and reflash the firmware to get rid of this problem.

wpa-2, 2024-12-24:
> Also, how can I fix the "Reading last session logs" issue? I’ve already tried `sudo rm -R /etc/pwnagotchi/log/*`, but it doesn’t work. I have to erase and reflash the firmware to get rid of this problem.

Check debug and read the wiki, My guess is something you have done is causing an issue. 
But that really isn't for git hub.