jayofelony/pwnagotchi #277: Flashing new release 2.9.3-2

scansonic, 2024-12-26:
After the flash and the first run, everything was tuned as usual. At the second start, nothing was normally functional.
Release: https://github.com/jayofelony/pwnagotchi/releases/download/v2.9.3-2/pwnagotchi-64bit.img.xz
Board: Raspberry PI 0 2W
Is it an error, or is it my stupidity?

Best regards, scansonic.
[image]
[image]
[image]

wpa-2, 2024-12-26:
Are per the instructions on the wiki you need to flash with the user PI only, once you reflash your issues should be sorted.

Linkatplug, 2024-12-26:
Don't put a custom user. You need to re-lash with the default user and change the password afterwards. It is possible to rename the home file, but this can generate errors later. I advise you to keep the user pi and change the password.

scansonic, 2024-12-26:
In the previous release, there was no such restriction. I changed the settings and everything worked fine.  I will go reflash and test as PI user.

Linkatplug, 2024-12-26:
> In the previous release, there was no such restriction. I changed the settings and everything worked fine.  I will go reflash and test as PI user.

This has been the case since the latest release. For previous versions, it had no impact.

wpa-2, 2024-12-26:
> > In the previous release, there was no such restriction. I changed the settings and everything worked fine.  I will go reflash and test as PI user.
> 
> This has been the case since the latest release. For previous versions, it had no impact.

actually that's not the case, you could have any user but things have changed recently.

jayofelony, 2024-12-26:
> In the previous release, there was no such restriction. I changed the settings and everything worked fine.  I will go reflash and test as PI user.

You dont need to reflash, just login as pi.

closed 2024-12-26 by jayofelony

scansonic, 2024-12-26:
> > In the previous release, there was no such restriction. I changed the settings and everything worked fine.  I will go reflash and test as PI user.
> 
> You dont need to reflash, just login as pi.

OK, I will try it.

scansonic, 2024-12-26:
> > In the previous release, there was no such restriction. I changed the settings and everything worked fine.  I will go reflash and test as PI user.
> 
> You dont need to reflash, just login as pi.

it didn't work out for me
[image]

wpa-2, 2024-12-26:
> > > In the previous release, there was no such restriction. I changed the settings and everything worked fine.  I will go reflash and test as PI user.
> > 
> > 
> > You dont need to reflash, just login as pi.
> 
> it didn't work out for me ![Screenshot 2024-12-26-5](https://private-user-images.githubusercontent.com/105582727/398746296-8329b1a4-ab7b-4e18-9a1e-1777b788f965.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyMjY2MjMsIm5iZiI6MTczNTIyNjMyMywicGF0aCI6Ii8xMDU1ODI3MjcvMzk4NzQ2Mjk2LTgzMjliMWE0LWFiN2ItNGUxOC05YTFlLTE3NzdiNzg4Zjk2NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyNlQxNTE4NDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNGVhYmM3OGVlYTM2YWIxYTEzNGZiM2Q3ZWEzYjNiZTA0ODkwZmE0OTk2ZjgyODlhNWM0NDljNzNkNGIyNDMzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.DNfpz4xq0i9YYJ9Il1ACE9tjQ7Gs9G7_SOJ_z-Q9TXY)
> 
> I will try to reflash it.

Check debug 

sudo pwnagotchi  --debug

scansonic, 2024-12-26:
I reflashed, and everything worked fine. So, I didn't need to enter a new user name, I just logged in as user PI.
Thank you all!

Best regards, scansonic.

wpa-2, 2024-12-26:
> I reflashed, and everything worked fine. So, I didn't need to enter a new user name, I just logged in as user PI. Thank you all!
> 
> Best regards, scansonic.

awesome happy pwning, can you close this as sorted then please.